PX4/PX4-Autopilot #10589: Uart Issue of PH4-mini with 1.8.0dev firmware

jamming, 2018-09-28:
The second uart(UART4_RX_I2C2 and UART4_TX_I2C2 pins) of PH4-mini should be multiplexed. By setting the parameter SYS_COMPANION to corresponding type, we can connect an osd, a SIK telemetry or a secondary GPS on it.

I have flashed the master firmware to PH4-mini via QGC on 27 Sep 2018. When I connecting an osd or a SIK telemetry to the second uart of PH4-mini, I found they don't work. I check SYS_COMPANION in parameter list, I found that it is disable by default. I try to change it to corresponding type and save, but the saving don't take effect. The parameter SYS_COMPANION  change back to disable after reboot. 

I try the beta firmware too. The parameter SYS_COMPANION can be change and saved. The second uart works well.

By the way, The parameter SYS_USE_IO is enable by default even there is no IO ship on  PH4-mini. To make the PWM OUTs working, I have to change it to disable manually. It is more convenient if the firmware recognize the FC type via rev or ver hardware, and set all parameters to correct type automatically.

hamishwillee, 2018-09-28:
I am working on the docs for this right now: https://github.com/PX4/px4_user_guide/pull/352

The summary is that SYS_COMPANION is no longer supported. Now you use a `*_CFG` or `*_CONFIG` parameter for a particular service or peripheral to set the port that it uses.

davids5, 2018-10-11:
In a resent PR SYS_USE_IO is set to 0 here https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L377 for the mini.

@jamming OK to close?

shortcut21, 2018-10-18:
Hello, guys!
I am trying to set up 115200 baud rate to Pixhawk 3 Pro and LairdTech 2.4ghz telemetry modules set at 115200 baud rate. I read that in the QGroundControl I need to set the sys_companion to 921600, but it's still not working. ( I am using the Firmware v1.8.0 stable) What should I do?

davids5, 2018-10-18:
> I am working on the docs for this right now: PX4/px4_user_guide#352

>The summary is that SYS_COMPANION is no longer supported. Now you use a *_CFG or *_CONFIG parameter for a particular service or peripheral to set the port that it uses.

@hamishwillee Is the the case for 1.8.0 stable?

hamishwillee, 2018-10-18:
> @hamishwillee Is the the case for 1.8.0 stable?

@davids5 No. The docs reflect master.

jinger26, 2018-12-03:
@dagar @davids5  this issue still exist for PH4mini with master according to @jamming

davids5, 2018-12-03:
@jinger26 - @dagar and I are unclear on what the issue is. The Schematic call the connector TELEM4, the product has a label that says UART & I2C B. Since there is not TELEM3 calling the port on the product TELEM 4 would not make sense.  @dagar is looking into changing the metadata that QGC uses to label it the same as on the product. 'UART & I2C B' We also need a way to tell QGC that the connectors on the 
V5 mini are a subset of the V5.

davids5, 2018-12-03:
@jinger26 Allso The https://docs.px4.io/en/assembly/quick_start_pixhawk4_mini.html has the wrong picture.  @hamishwillee

hamishwillee, 2018-12-03:
THanks @davids5 . @jinger26 This is going to have to come from you - I don't have any other imagery.

jamming, 2018-12-04:
UART&I2CB should be a multipurpose  port,  we can connect different peripherals(Telemetry, Additional GPS, OSD, etc ) on it.

UART Mapping example for PH4-mini
 QGC side    CPU side
•SERIAL0 -> USB
•SERIAL1 -> UART2 (Telem1)
•SERIAL2 -> UART4 (Telem2, Additional GPS, OSD, etc )
•SERIAL3 -> UART1 (GPS)
•SERIAL4 -> UART6 (RCIN port)
•SERIAL5 -> UART7 (debug)
The Telem1 port has RTS/CTS pins, the other UARTs do not have RTS/CTS.

PH4-mini   pinout
[https://github.com/jamming/image/blob/master/PH4-MINI.png](url)

davids5, 2018-12-04:
@hamishwillee, @jamming provided some images in the link above. Will that work?

hamishwillee, 2018-12-04:
@davids5 No, but the PR from @pfjare did: https://github.com/PX4/px4_user_guide/pull/396 :-)

Fixed: https://docs.px4.io/en/flight_controller/pixhawk4_mini.html

jinger26, 2018-12-06:
Rewriting Holybro's feedback here:

If "UART&I2CB" is not configurable, then the port loses its original design purpose  (to enable multipurpose functionalities that could be set by users, i.e. dual GPS); right now the firmware does not assign default value to UART4, nor have a parameter that users can access to configure in QGC. 

Also parameters not in sync with doc parameter reference page (https://dev.px4.io/en/advanced/parameter_reference.html). The doc reflects master, and Pixhawk 4/mini are shipped with lastest stable, I think @hamishwillee has a WIP PR on this version control; and @LorenzMeier mentioned on the dev-call to file a request under QGC for a robust caching/storage system for all parameters.

hamishwillee, 2018-12-07:
> , I think @hamishwillee has a WIP PR on this version control;

I have an [issue](https://github.com/PX4/Devguide/issues/676) that specifies what I need. The work has to be done in Jenkins by @dagar. Essentially this will allow us to build (and cross link) complete doc sets for each release. I do not know when this is scheduled.

But this will not automatically result in docs builds for older releases. I took a tag 10 days ago https://github.com/PX4/Devguide/releases/tag/v1.8.2. If that doesn't have the correct parameters "current stable parameters"  then someone will need to find the docs version that has them and we could perhaps take a branch from that. 
Note also that I don't plan to update past docs with new improvements by default. So fixes to the images etc that went into master after a branch might not appear in the old version of the docs. I can address this on a case by case basis - but I'd prefer not to do much because there is more than enough work to do.

Note that whatever it ships with, our expectation is that people would update to newer firmware versions.

closed 2019-07-08 by stale[bot]

petertorelli, 2021-07-15:
Is there a way to configure PX4mini `uart&i2cb` yet? The [reference page](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#GPS_2_CONFIG) indicates it is now called TELEM/SERIAL4, which is consistent with @jamming and @davids5 comments above. But there is still no reading in `nsh` when selected TELEM/SERIAL4 aka `/dev/ttyS3` (below). There was a comment about it being multiplexed and some discussion about how to configure it for UART or I2C but the companion parameter no longer exists and there was no specifics about the new variable in this thread.

What's the current status on enabling this, or is it called something else now?

```
NuttShell (NSH) NuttX-10.0.0
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyS0, baudrate: 115200
INFO  [gps] sat info: disabled
INFO  [gps] rate reading:            697 B/s
INFO  [gps] rate position:           5.00 Hz
INFO  [gps] rate velocity:           5.00 Hz
INFO  [gps] rate publication:          5.00 Hz
INFO  [gps] rate RTCM injection:      0.00 Hz
 sensor_gps_s
    timestamp: 20126812  (0.027335 seconds ago)
    time_utc_usec: 1626310949800314
    device_id: 11010053 (Type: 0xA8, SERIAL:0 (0x00)) 
    lat: 454650867
    lon: -1227407165
    alt: 119270
    alt_ellipsoid: 98056
    s_variance_m_s: 0.3540
    c_variance_rad: 1.8132
    eph: 1.5130
    epv: 1.9880
    hdop: 0.8000
    vdop: 0.8700
    noise_per_ms: 117
    jamming_indicator: 8
    vel_m_s: 0.0220
    vel_n_m_s: 0.0220
    vel_e_m_s: -0.0040
    vel_d_m_s: -0.0380
    cog_rad: 0.0000
    timestamp_time_relative: 0
    heading: nan
    heading_offset: 0.0000
    automatic_gain_control: 2106
    fix_type: 3
    jamming_state: 0
    vel_ned_valid: True
    satellites_used: 14
INFO  [gps] 
INFO  [gps] Secondary GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: NOT OK, port: /dev/ttyS3, baudrate: 57600
INFO  [gps] sat info: disabled
INFO  [gps] rate reading:              0 B/s
nsh> 
```

jamming, 2021-07-15:
PX4mini 标识为UART&amp;I2CB的端口，在固件里面对应的确实是TELEM/SERIAL4. 这个标识差异是丝印设计和固件设计时没有充分沟通。

 
 这个UART&amp;I2CB是可配置的，可以作为第二TELEM,OSD,第二GPS

如果作为第二GPS,参考设置如下：
1）把GPS2 MAP到TELEM/SERIAL4

2）在Parameters-&gt;Serial把 SERIAL4 波特率设为AUTO

重启之后，接上第二GPS到标识为UART&amp;I2CB的端口，应该就能看到第二GPS识别出来了。

 NuttShell (NSH) NuttX-10.0.0 nsh&gt; gps status INFO  [gps] Main GPS INFO  [gps] protocol: UBX INFO  [gps] status: OK, port: /dev/ttyS0, baudrate: 115200 INFO  [gps] sat info: disabled INFO  [gps] rate reading:            697 B/s INFO  [gps] rate position:           5.00 Hz INFO  [gps] rate velocity:           5.00 Hz INFO  [gps] rate publication:          5.00 Hz INFO  [gps] rate RTCM injection:      0.00 Hz  sensor_gps_s     timestamp: 20126812  (0.027335 seconds ago)     time_utc_usec: 1626310949800314     device_id: 11010053 (Type: 0xA8, SERIAL:0 (0x00))      lat: 454650867     lon: -1227407165     alt: 119270     alt_ellipsoid: 98056     s_variance_m_s: 0.3540     c_variance_rad: 1.8132     eph: 1.5130     epv: 1.9880     hdop: 0.8000     vdop: 0.8700     noise_per_ms: 117     jamming_indicator: 8     vel_m_s: 0.0220     vel_n_m_s: 0.0220     vel_e_m_s: -0.0040     vel_d_m_s: -0.0380     cog_rad: 0.0000     timestamp_time_relative: 0     heading: nan     heading_offset: 0.0000     automatic_gain_control: 2106     fix_type: 3     jamming_state: 0     vel_ned_valid: True     satellites_used: 14 INFO  [gps]  INFO  [gps] Secondary GPS INFO  [gps] protocol: UBX INFO  [gps] status: NOT OK, port: /dev/ttyS3, baudrate: 57600 INFO  [gps] sat info: disabled INFO  [gps] rate reading:              0 B/s nsh&gt;

petertorelli, 2021-07-15:
@jamming ... /runs google translate/

QGroundControl doesn't expose SERIAL4 settings in the Parameters tab, how can I set this via `nsh`?

davids5, 2021-07-15:
Translation 

PX4mini is marked as the port of UART&amp;I2CB, and the corresponding port in the firmware is indeed TELEM/SERIAL4. The difference between this logo is that the silk screen design and the firmware design are not fully communicated.

&nbsp;
&nbsp;This UART&amp;I2CB is configurable and can be used as the second TELEM, OSD, and second GPS

**If used as the second GPS, the reference settings are as follows:
1) MAP GPS2 to TELEM/SERIAL4

2) Set the SERIAL4 baud rate to AUTO in Parameters->Serial**

After restarting, connect the second GPS to the port labeled UART&amp;I2CB, and you should be able to see the second GPS recognized.

petertorelli, 2021-07-15:
@davids5 
Hi David, thanks. I mapped GPS 2o TELEM/SERIAL4 on QGroundControl's parameter tab, but there is no SERIAL4 in Parameters->Serial, it only goes up to SERIAL2. Is there a way to change this from `nsh`? Or different configuration software?
Thanks.

jamming, 2021-07-16:
> @davids5
>  Hi David, thanks. I mapped GPS to TELEM/SERIAL4 on QGroundControl's parameter tab, but there is no SERIAL4 in Parameters-&gt;Serial, it only goes up to SERIAL2. Is there a way to change this from nsh? Or different configuration software?
>  Thanks.

What version is your firmware?  I test this with 1.12.0 stable, It works well.

petertorelli, 2021-07-16:
I edited my last reply, probably should have commented: it all works fine now, I mistakenly changed things in the wrong order. Thanks!