PX4/PX4-Autopilot #10607: DriverFramework merge with in tree PX4 drivers

dagar, 2018-09-30:
With PX4 SPI and I2C support extended to Linux (i2c-dev and spidev) and qurt we can look at extending the existing in tree NuttX drivers to support Linux and Qurt platforms. Eventually we'll no longer need DriverFramework or df wrappers.

PX4 drivers available
 * df_mpu9250_wrapper
 * df_bmp280_wrapper
 * df_ms5607_wrapper
 * df_ms5611_wrapper
 * df_mpu6050_wrapper
 * df_ak8963_wrapper
 * df_hmc5883_wrapper
 * df_trone_wrapper

Missing in PX4 (only in DriverFramework)
 * df_bebop_bus_wrapper
 * df_bebop_rangefinder_wrapper
 * df_isl29501_wrapper
 * df_lsm9ds1_wrapper
 * df_ltc2946_wrapper

dagar, 2018-09-30:
FYI @thomasgubler

mrpollo, 2018-11-13:
@dagar who is doing the work to migrate the missing wrappers?

dagar, 2018-11-13:
This larger effort to rewrite the mpu9250 driver is being done in such a way that it will work on linux as well. https://github.com/PX4/Firmware/pull/10602

Once we have complete linux/qurt support for in tree drivers the DriverFramework versions (wrappers and submodule) can simply go away.

dagar, 2019-06-24:
Slowly coming along. https://github.com/PX4/Firmware/pull/12136

dagar, 2020-04-08:
Done.

closed 2020-04-08 by dagar