PX4/PX4-Autopilot #10610: Minor - tune_control command line reports 1-15, while there appears to be 19?

igalloway, 2018-09-30:
**Describe the bug**
This is a minor detail in self documenting

The tune _control command indicates 
    -t <defualt tunes>      Play the default (1...15) (default=1)
when there are more than that,  i believe 19?

**To Reproduce**
at the console manually issue the "tone_control"
The output will say "Play the default (1...15)"

**Expected behavior**
Mention the actual number of default tones, or eliminate the mention of "15"
Play the default (1...19)
Play the default (1...XX)  

**Log Files and Screenshots**
tune_control play [-t <default tunes>] [-f <frequency>] [-d <duration>] [-s <strength>] [-m < melody>]

        -t <defualt tunes>      Play the default (1...15) (default=1)
        -f <frequency>          Frequency from 0-20kHz
        -d <duration>           Duration of the tone in us
        -s <strength>           Strength of the tone between 0-100
        -m <melody>             Melody in a string form ex: "MFT200e8a8a"

tune_control stop               Stops playback, useful for repeated tunes

hamishwillee, 2019-02-04:
This is still true. Ran into while generating sounds for https://github.com/PX4/px4_user_guide/pull/431

FYI @davids5

hamishwillee, 2019-06-27:
Still true

davids5, 2019-06-27:
@hamishwillee 

Looks like it was fixed
```
nsh> tune_control
Usage: tune_control <command> [arguments...]
 Commands:

   play          Play system tune or single note.
     [-t <val>]  Play predefined system tune
                 default: 1
     [-f <val>]  Frequency of note in Hz (0-22kHz)
     [-d <val>]  Duration of note in us
     [-s <val>]  Volume level (loudness) of the note (0-100)
                 default: 40
     [-m <val>]  Melody in string form
                 values: <string> - e.g. "MFT200e8a8a"

   libtest       Test library

   stop          Stop playback (use for repeated tunes)
nsh>
```

closed 2019-06-27 by davids5