PX4/PX4-Autopilot #10614: Inconsistent magnetonometer: External GPS 2 inconsistent unless yaw of 180 relative to GPS 1

zanemountcastle, 2018-09-30:
**Describe the bug**
When using a Pixhawk 2.1 flight controller, QGroundControl complains about an inconsistency in the incoming magnetometer data. It turns out that the way to satisfy QGC (and PX4) is to set the orientation of GPS2 (an external 3DR GPS module) to a yaw of 180. It seems as though PX4 rotates the incoming magnetometer data and therefore doesn't recognize that both modules are pointing in the same direction even though they are. 

**To Reproduce**
Steps to reproduce the behavior:
1. Disable the onboard mag. (Not sure that this matters, but this is how I have it for other reasons.)
2. Configure two external GPS modules pointing forward on the airframe.
3. Connect to the Pixhawk in QGroundControl.
4. See MAG SENSOR INCONSISTENT error preventing arming.

**Expected behavior**
Rotating the orientation of the GPS module in software should not have to happen since it's obviously a misleading error. PX4 should recognize that both mags are oriented in the same direction. 

**Log Files and Screenshots**
n/a

**Drone (please complete the following information):**
PX4 "Standard Plane" v1.8 firmware.

**Additional context**
PX4 did not complain about this before v1.8.

dagar, 2018-09-30:
Have you done a full mag calibration? If so can you share the CAL_MAG_* parameters? From the rotation parameter we'll be able to see if PX4 thinks that mag is internal or external.

zanemountcastle, 2018-10-01:
Here are the CAL_MAG parameters:

```
1	1	CAL_MAG0_EN	1	6
1	1	CAL_MAG0_ID	73225	6
1	1	CAL_MAG0_ROT	0	6
1	1	CAL_MAG0_XOFF	0.091849252581596375	9
1	1	CAL_MAG0_XSCALE	1.016670107841491699	9
1	1	CAL_MAG0_YOFF	0.027895305305719376	9
1	1	CAL_MAG0_YSCALE	0.920327067375183105	9
1	1	CAL_MAG0_ZOFF	-0.085807152092456818	9
1	1	CAL_MAG0_ZSCALE	1.078429102897644043	9
1	1	CAL_MAG1_EN	1	6
1	1	CAL_MAG1_ID	73233	6
1	1	CAL_MAG1_ROT	4	6
1	1	CAL_MAG1_XOFF	-0.046974591910839081	9
1	1	CAL_MAG1_XSCALE	1.014231324195861816	9
1	1	CAL_MAG1_YOFF	-0.038735963404178619	9
1	1	CAL_MAG1_YSCALE	0.923173666000366211	9
1	1	CAL_MAG1_ZOFF	-0.165157869458198547	9
1	1	CAL_MAG1_ZSCALE	1.074820637702941895	9
1	1	CAL_MAG2_EN	0	6
1	1	CAL_MAG2_ID	263202	6
1	1	CAL_MAG2_ROT	-1	6
1	1	CAL_MAG2_XOFF	-0.094515621662139893	9
1	1	CAL_MAG2_XSCALE	1.003441929817199707	9
1	1	CAL_MAG2_YOFF	0.256852358579635620	9
1	1	CAL_MAG2_YSCALE	0.987532377243041992	9
1	1	CAL_MAG2_ZOFF	0.218661323189735413	9
1	1	CAL_MAG2_ZSCALE	1.014442443847656250	9
1	1	CAL_MAG3_EN	0	6
1	1	CAL_MAG3_ID	131618	6
1	1	CAL_MAG3_ROT	-1	6
1	1	CAL_MAG3_XOFF	-0.041170552372932434	9
1	1	CAL_MAG3_XSCALE	0.974972546100616455	9
1	1	CAL_MAG3_YOFF	0.104813583195209503	9
1	1	CAL_MAG3_YSCALE	0.992948651313781738	9
1	1	CAL_MAG3_ZOFF	-0.003898021299391985	9
1	1	CAL_MAG3_ZSCALE	1.036365270614624023	9
1	1	CAL_MAG_PRIME	73225	6
1	1	CAL_MAG_SIDES	63	6
```

I've attached the full parameter file as well: [params.txt](https://github.com/PX4/Firmware/files/2435255/params.txt)