PX4/PX4-Autopilot #10619: Cygwin: Gazebo Support

MaEtUgR, 2018-10-01:
This is a bigger undertaking but the goal is to support gazebo simulation when running PX4 SITL on Windows natively.

The first and probably biggest problem is that [osrf](https://www.openrobotics.org/) does not provide any official Windows binary for gazebo. I also didn't find any third party build so it's clear we need to build from source even for the first test.

A clue is this tutorial on the official project page http://gazebosim.org/tutorials?tut=install_on_windows

Status
By now I started to setup the whole environment, download and build all dependencies listed in the tutorial. After a lot of trial and error they build successfully and I went on to try and run cmake for gazebo. It hangs with a lot of errors, it doesn't even recognize all the necessary boost libraries...

I have to continue from there and will post updates. If you want to participate and need any instructions for what I have so far please don't hesitate to contact me.

FYI

dagar, 2018-10-01:
What about using the Windows Linux subsystem? https://answers.ros.org/question/238646/installing-ros-on-ubuntu-bash-in-windows-10/

Ubuntu is in the Windows 10 store. https://www.microsoft.com/en-us/p/ubuntu/9nblggh4msv6#activetab=pivot:overviewtab

MaEtUgR, 2018-10-02:
@dagar I'm still concerned about it just being a virtual machine which is from system resources like graphics perspective independent of Windows. But I might give it a try... just to have some experience.

TSC21, 2019-01-20:
@MaEtUgR we do want to keep this as relevant right?

MaEtUgR, 2019-01-22:
@TSC21 Yes but I shifted the matter to lower priority since I realized it will consume a lot of time.
1. Compile gazebo for Windows natively (Visual C++ compiler) I tried to do so following http://gazebosim.org/tutorials?tut=install_on_windows but either the instructions are not complete or outdated because I ran into all sorts of problems. It seems that this builds on a couple of prebuilt binaries and a hacked cmake setup which could in theory work but it's hard to tell how far you got so far and if the resulting compilation will work as expected. From a preformance standpoint it's definitely the prefered way.
2. Install gazebo on the Windows Linux subsystem. I did some tests with Ubuntu running on WLS and in terms of contained CPU performance it seems awsome, definitely no VM but the posix applications run directly on the Windows kernel. But the problem with that approach is I/O: There isn't any graphics support which means you need to either have a gazebo client running on the Windows side or run an xserver and access it with some not very well supported xserver network client for Windows (which is tedious). On the positive side this seems to be relatively well feasible (with xserver over network) but I doubt the graphics performance will be usable. I could do a test with: https://dev.px4.io/en/setup/dev_env_windows_bash_on_win.html Another problem is the easy reproducability since WLS and Ubuntu base system need to be installed manually from GUI and none of this is directly compatible with the Cygwin toolchain or its setup.

hamishwillee, 2019-06-27:
@MaEtUgR I'd still love this. However I think it might be easier in near future. Tully at the dev summit indicated that Gazebo "Ignition" (re-written from ground-up version) should run on Windows out of the box.

MaEtUgR, 2019-06-27:
@hamishwillee I totally agree. I definitely don't have the bandwidth to port the original hacky gazebo for windows setup but the discussion with Tully at the Dev Summit was indeed very educational and I'm looking forward to test the new ignition library and hopefully even ROS2 on Windows here are the latest gazebo instructions for Linux, there's no out of the box Windows binary as far as I can see but I guess that will come: https://ignitionrobotics.org/libs/gazebo

MaEtUgR, 2019-10-06:
Since I don't see how I will find any time to work on this and plan to rely on https://ignitionrobotics.org/libs/gazebo getting Windows support I'm closing this issue.

closed 2019-10-06 by MaEtUgR