PX4/PX4-Autopilot #10620: pixhawk 2.1 voltaje issues.

Avysuarez, 2018-10-01:
I had problems with pixhawk 2.1 voltage reading,  I have calibrated the voltage several times  and even with the full battery, activate the fail safe immediately. 
I already tested with another voltage sensor and reset the parameters and it's still the same.

https://review.px4.io/plot_app?log=8c1f6603-3183-4bf6-87fd-6d527988a276
https://review.px4.io/plot_app?log=23ce4f5f-e26c-4a00-96e7-5e7cf5c8138c

dagar, 2018-10-31:
Obviously there isn't much to go on, but can you describe the full flight, the fly away, and post the last few logs from this vehicle that you do have?

Did the flyaway correspond with telemetry disconnecting?
Did the vehicle look stabilized?
What was the exact build you were testing?

Tony3dr, 2018-10-31:
@dagar I will gather the data and details and send it over to you. Thanks.

Avysuarez, 2018-10-31:
**Setup of the build.** 
DJI F450
Pixhawk 2.1 (cube)
GPS here +
Motors Tmotors 2212 920kv
ESC DJI 420 lite 20A
Frsky D4r-II
Holybro 433 telemetry radio

The copter had two successful flight before the flyaway. On the flight of the flyaway, upon the takeoff and during the mission, the copter was fly stable, we lost sight of it for a moment and we did not see it again. 
Please note- The copter was not connected to the telemetry that is the reason why we didn't find it. We were having an issue connecting to the telemetry so we decided to fly it without telemetry.

The commit of this flight was the 44f650f.
This is one of the last logs that was made with this copter.
https://review.px4.io/plot_app?log=a8246a6a-d917-495a-ad5e-3295b9e93c02

closed 2019-01-30 by Tony3dr