PX4/PX4-Autopilot #10621: Magnetometer Calibration seems not possible for external MAG on CAN Bus

Seeelefant, 2018-10-02:
**Describe the bug**
I always have inconsistent magnetometers, despite many calibration dances

Looking at the parameter settings there are no calibration settings for the external mag
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

Seeelefant, 2018-10-02:
[image]

[image]

Logs can be found here: https://review.px4.io/plot_app?log=d2d179c5-1404-49b6-9032-8528c4dfefba

mhkabir, 2018-10-02:
Probably something bug/regression in the UAVCAN magnetometer interface, which has existed for some time. I remember this exact issue from a few months back.

Seeelefant, 2018-10-02:
Maybe this one https://github.com/PX4/Firmware/issues/7395 ? Sounds that this will be a longer journey :-(

Seeelefant, 2018-10-03:
It´s definitively a bug, with ArduPilot it´s working :-( .

LorenzMeier, 2018-11-30:
@bkueng @RomanBapst

nrodichenko, 2018-12-11:
I've come across what seems to be the same issue. I can calibrate external mag on CAN bus, but would see the same lack of CAL_MAG1* parameters after reboot (only _ID and _ROT), and "mag sensors inconsistent error".

I can see the CAL_MAG1* parameters after calibration without reboot, they all have assigned indices and values. After reboot the parameter values are still there but indices are all "-1".

I've been able to narrow down the problem to a race condition between the UAVCAN magnetometer init and start of the "sensors" app. By moving UAVCAN init before rc.senors init in rcS, and introducing a significant delay (sleep 6000000), I'm able to achieve correct CAN mag startup. I believe this is connected to boot delay in GNSS2 boot loader (approx. 5 seconds as per data sheet). 

Another way to circumvent the issue without introducing delay is to put UAVCAN init before rc.sensors, start the autopilot, and issue a reboot command via CLI. In this case GNSS2 doesn't enter boot loader and race condition is met.

I would be glad to work on fixing the issue, but would appreciate some intro to sensor enumeration mechanisms by sensors app.

HW: Zubax GNSS2 with a custom FMUV4-based autopilot (single onboard mag)
FW: 1.8.2 with minor adjustments not related to UAVCAN

RavWin, 2018-12-11:
> I've come across what seems to be the same issue. I can calibrate external mag on CAN bus (Zubax GNSS2), but would see the same lack of CAL_MAG1* parameters after reboot (only _ID and _ROT), and "mag sensors inconsistent" error.
> 
> I've been able to narrow down the problem to a race condition between the UAVCAN magnetometer init and start of the "sensors" app. By moving UAVCAN init before rc.senors init in rcS, and introducing a significant delay (usleep 6000000), I'm able to achieve correct CAN mag startup. I believe this is connected to boot delay in GNSS2 bootloader (approx. 5 seconds as per datasheet).
> 
> Another way to circumvent the issue without introducing delay is to put UAVCAN init before rc.sensors, start the autopilot, wait no less than 5 seconds, and issue a reboot command via CLI. In this case GNSS2 doesn't enter bootloader and race condition is met.
> 
> I would be glad to work on fixing the issue, and would appreciate if someone can point me in the right direction with sensor enumeration mechanisms by sensors app.
> 
> HW: Zubax GNSS2 with a custom FMUV4-based autopilot (single onboard mag)
> FW: 1.8.2 with minor adjustments not related to UAVCAN

I've posted a hotfix for this error, but it doesn't fix the problem itself in my case. Just the calibration bug.
But the rotating MAG with UAVCAN main sensor is still there.

dagar, 2018-12-11:
@RavWin would you like to open a PR with your hotfix?

RavWin, 2018-12-11:
> @RavWin would you like to open a PR with your hotfix?

If it is working, I'll make a pull request. Could you test it?

nrodichenko, 2018-12-12:
I can test fly it by Friday.

mhkabir, 2018-12-12:
I think I remember this race condition from sometime back. Let me look at what we did back then to fix it.

RavWin, 2018-12-12:
@mhkabir, did you mention v1.6.2 mag fix, when mag rotation didn't apply in voted_sensors_update on reboot?

nrodichenko, 2019-01-04:
@mhkabir any updates on this one?

@RavWin ultimately didn't test your branch since it only deals with calibration (which was working for me in 1.8.2 stable), and not the startup initialization.

Seeelefant, 2019-01-05:
Good to hear that, from which release on is this fix available?Is it on the roadmap for 1.9?

RavWin, 2019-01-05:
> Good to hear that, from which release on is this fix available?Is it on the roadmap for 1/9&

I don't know. I've made this fix by myself. I'm not in PX4 developer team, just a side developer.

Seeelefant, 2019-01-05:
So the question goes to @LorenzMeier .

Juanelm, 2019-02-18:
Following up on this. Im getting the same issue every time I boot up.  Will the fix be on 1.9?  @LorenzMeier

julianoes, 2019-07-11:
Is this still an issue?

RavWin, 2019-07-11:
> Is this still an issue?

Yes, cause my patch wasn't applied

julianoes, 2019-07-15:
@RavWin Ok thanks for the note. Is there a pull request we should look at?

tstellanova, 2019-10-04:
Encountering this same problem with a Pixhawk2, trying two different GPS+mag modules:

- Zubax GNSS2 (uavcan)
- Drotek DP0302 (NEO-M8T GPS + LIS3MDL Magnetometer)

Went through the compass calibration successfully with both GPS+mag modules multiple times.  However, when we attempt to arm, we get the error "Preflight Fail: Compass sensors inconsistent"

julianoes, 2020-02-12:
@dagar can we determine which external mag we have and if we have to rotate it for the Drotek case?

RavWin, 2020-02-12:
> @dagar can we determine which external mag we have and if we have to rotate it for the Drotek case?

The ID of mag is unique and we can use it to determine which mag we have

julianoes, 2020-02-12:
@RavWin nice! If you could create a PR to fix it, that would be fantastic.

RavWin, 2020-02-12:
> @RavWin nice! If you could create a PR to fix it, that would be fantastic.

I've never worked with Drotek mag, but my own can Mag uses the same logic inside and I've made a PR for mag initialization fix a year ago but it was not concerning orientation but calibration issue. As far as I know it wasn't applied.

julianoes, 2020-02-14:
@RavWin ok what was that PR? Sorry it got dropped. Could you update it?

dagar, 2020-02-17:
> @dagar can we determine which external mag we have and if we have to rotate it for the Drotek case?

It depends on the specifics. By convention going back to the original 3DR compass/mag the mag was mounted weirdly relative to the printed arrow on the unit.

https://github.com/PX4/Firmware/blob/86bf4b2289c1630833c920c84dfc95a82e2ac6b5/src/drivers/magnetometer/hmc5883/HMC5883.cpp#L615-L620

Years later (going beyond the hmc5883 and 3DR) some units copy this, some don't. It's actually pretty terrible.

Roman-, 2020-03-04:
I will add my two cents in here.

The UavcanMagnetometerBridge seems to be designed to support multiple magnitometers connected through uavcan. That is explicitly claimed [there](https://github.com/PX4/Firmware/blob/2b087a92c277d542405c79fd98e99e5ccfb358ca/src/drivers/uavcan/sensors/sensor_bridge.hpp#L85). However the instance of UavcanMagnetometerBridge [only has one mag_calibration_s structure](https://github.com/PX4/Firmware/blob/2b087a92c277d542405c79fd98e99e5ccfb358ca/src/drivers/uavcan/sensors/mag.hpp#L78), and there's only one instance of UavcanMagnetometerBridge created ([there](https://github.com/PX4/Firmware/blob/2b087a92c277d542405c79fd98e99e5ccfb358ca/src/drivers/uavcan/sensors/sensor_bridge.cpp#L53)).

So the UavcanMagnetometerBridge can only store one mag calibration data at once, which makes multiple-mag-support-design useless. Am I missing something? @pavel-kirienko

pavel-kirienko, 2020-03-05:
> So the UavcanMagnetometerBridge can only store one mag calibration data at once, which makes multiple-mag-support-design useless.

@TSC21 can we please register this to have it fixed in the v1 rewrite? Not sure what the proper workflow would be.

junwoo091400, 2021-12-18:
Has this been solved @pavel-kirienko ? @RavWin