PX4/PX4-Autopilot #10623: Issue with input capture timestamp

DanielePettenuzzo, 2018-10-02:
I'm testing the camera capture driver PR (https://github.com/PX4/Firmware/pull/10295) and I'm having some issues with the timestamp of the capture. I am generating an active high pulse from my pixhawk and looping it back into the input capture. The capture driver is set to capture falling edges. Sometimes my trigger falling edge and the capture timestamp match and sometimes I get a 20ms delay between the two.

Using a 4ms active high pulse as trigger:

- When there is a 20 ms delay
trigger (rising edge):   76723791 us
trigger (falling edge):   76727501 us
capture (falling edge): 76747500 us

- When it works
trigger (rising edge): 160397157 us
trigger (falling edge): 160401164 us
capture (falling edge): 160401164 us

Moreover in my final setup I would like to be able to able to capture the falling edge of a 100us active high pulse. Is this possible with the current capture setup?

DanielePettenuzzo, 2018-10-02:
@davids5 FYI

davids5, 2018-10-02:
@DanielePettenuzzo would you please provide a link to the test code and any setup info.

davids5, 2018-10-02:
@DanielePettenuzzo - What HW? What branch? Any special test code? Please provide a link and include the pins you are connected to.

davids5, 2018-10-02:
I ran a quick test.

FMUv5 Mini
Tied FMU_6 -  FMU_5

```
camera_trigger pwm (auto started)
camera_capture start
camera_capture on
```
Params
```
x   TRIG_ACT_TIME [609,1165] : 40.0000
x   TRIG_DISTANCE [610,1166] : 25.0000
x   TRIG_INTERFACE [611,1167] : 4
x   TRIG_INTERVAL [612,1168] : 40.0000
x + TRIG_MODE [613,1169] : 2
x + TRIG_PINS [614,1170] : 5
    TRIG_POLARITY [-1,1171] : 0

    CAM_CAP_DELAY [-1,117] : 1.0000
  + CAM_CAP_EDGE [-1,118] : 1
x   CAM_CAP_FBACK [44,119] : 0
    CAM_CAP_MODE [-1,120] : 0
```
Code to do diff:
```
diff --git a/src/drivers/camera_capture/camera_capture.cpp b/src/drivers/camera_capture/camera_capture.cpp
index 8aed49cbef..e247adbfe1 100644
--- a/src/drivers/camera_capture/camera_capture.cpp
+++ b/src/drivers/camera_capture/camera_capture.cpp
@@ -86,6 +86,10 @@ CameraCapture::~CameraCapture()
        camera_capture::g_camera_capture = nullptr;
 }
 
+uint32_t ov[4096];
+uint32_t edge_times[4096];
+int32_t edge_index = -1;
+
 void
 CameraCapture::capture_callback(uint32_t chan_index,
                                hrt_abstime edge_time, uint32_t edge_state, uint32_t overflow)
@@ -94,6 +98,17 @@ CameraCapture::capture_callback(uint32_t chan_index,
        struct _trig_s trigger;
 
        trigger.chan_index = chan_index;
+  if (edge_index < 4096) {
+      edge_index++;
+      edge_times[edge_index] = edge_time;
+      ov[edge_index] = overflow;
+  }

        trigger.edge_time = edge_time;
        trigger.edge_state = edge_state;
        trigger.overflow = overflow;
@@ -117,6 +132,14 @@ CameraCapture::publish_trigger()
 {
        struct _trig_s trig;
 
+       if (edge_index == 4096)
+         {
+           edge_index++;
+           for (int i = 4095; i > 0; i -= 2) {
+               printf("%d %d %d\n", edge_times[i-1], edge_times[i], edge_times[i] - edge_times[i-1]);
+           }
+         }
+
```
data:
Run time 82 Seconds.
[... 2029 lines cut ...]
24926979 | 24946978 | 19999 | -1 |   |   |   |  
24886979 | 24906979 | 20000 | 0 |   |   |   |  
24846979 | 24866978 | 19999 | -1 |   |   |   |  
24806979 | 24826979 | 20000 | 0 |   |   |   |  
24766979 | 24786978 | 19999 | -1 |   |   |   |  
24726978 | 24746979 | 20001 | 1 |   |   |   |  
24686979 | 24706979 | 20000 | 0 |   |   |   |  
24646978 | 24666979 | 20001 | 1 |   |   |   |  
24606979 | 24626979 | 20000 | 0 |   |   |   |  
24566979 | 24586979 | 20000 | 0 |   |   |   |  
24526978 | 24546979 | 20001 | 1 |   |   |   |  
24486978 | 24506979 | 20001 | 1 |   |   |   |  
24446979 | 24466979 | 20000 | 0 |   |   |   |  
24406978 | 24426978 | 20000 | 0 |   |   |   |  
24366979 | 24386979 | 20000 | 0 |   |   |   |  
24326978 | 24346979 | 20001 | 1 |   |   |   |  
24286978 | 24306979 | 20001 | 1 |   |   |   |  
24246978 | 24266979 | 20001 | 1 |   |   |   |  
24206978 | 24226979 | 20001 | 1 |   |   |   |  
24166979 | 24186979 | 20000 | 0 |   |   |   |  
24126979 | 24146978 | 19999 | -1 |   |   |   |  
24086978 | 24106979 | 20001 | 1 |   |   |   |  
24046978 | 24066979 | 20001 | 1 |   |   |   |  
24006978 | 24026979 | 20001 | 1 |   |   |   |  
23980796 | 23986978 | 6182 | -3928 |   |   |   |

davids5, 2018-10-02:
@DanielePettenuzzo - I did a test run where I hacked the timer to 100 us with a 257us period.
I have +-1 us (the total of 4097 deltas is 13 us)

81967581 | 81967838 | 257 | 0
-- | -- | -- | --
81967067 | 81967324 | 257 | 0
81966552 | 81966810 | 258 | 1

Please describe what you are using as the source? 
GPIO would explain it.

DanielePettenuzzo, 2018-10-03:
@davids5 Thank you very much for looking at this. I am using an fmu-v5 (pixhawk 4) on the pr branch. I basically added a printf after toggling the trigger pin and one in the camera_capture to print the capture time. There is no special code added and I'm using fmu pin 2 to trigger and fmu pin 6 to capture. 

I am using the GPIO trigger because I could easily print the time of the rising and falling edge. What is the issue with this configuration? The delay I'm seeing is caused by this?

closed 2019-01-20 by davids5