PX4/PX4-Autopilot #10624: Vehicle beeps with no SD card

dakejahl, 2018-10-02:
**Describe the bug**
Vehicle beeps at you continuously if there is no SD card inserted. 

**Expected behavior**
No beeping.

mhkabir, 2018-10-02:
I can see why beeping it good for most use cases, but we can add a circuit breaker.

igalloway, 2018-10-05:
> It boots it just doesn't stop beeping.

I think the beeping is a good idea, but agree that a circuit breaker should
be added.
However what i found very confusing was trying to figure out what this was
initially.
Is there a way for this to show as a warning in QGC? I was expecting to see
something there when i heard the beeping.

dakejahl, 2018-10-09:
FYI @davids5 
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L146
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L169

These make the vehicle beep until the end of time. Should we just create a circuit breaker parameter and wrap these in an `if param compare SD_BEEP_CBRK`?

davids5, 2018-10-09:
@dakejahl Good question. The change was intentional restoring how it worked in the past that made sense: Indicate a fatal boot issues. 

There is a command that can be issued `tune_control stop` that will stop the beep.

I think there is more to think about here:

Is the scope just SD card beep or any beep for that matter? 

There are dependencies on having a working system (in this case an SD card). For Instance UAVCAN needs a device store node information.  The question is "Is **this** craft flight ready?"  If not, how do you tell the user something is wrong and when can you tell them. 

The board had an issue in the boot sequences. The Answer is NO it is not ready for flight and there is a problem that need to be investigated and resolved.
 
The indications can be sound, lights and a UI. The ability of the system to indicate a failure during boot, increases and the boot progresses.
 
If there is a high level UI connected and a User can acknowledge an error, and choose and option to 1 ) ignore a Failed SD card or 2) ignore it permanently that would ensure the TOP decision maker has input. But what happens when that  decision maker then makes another change that this one affects and does not know it?  

We need rules and method for escalation that work. The circuit breakers solve one problem and creates another. 

The Manifest can tell us what HW is on this board and is expected to work vrs HW that has failed. We should use this and a loss-less escalation method to notify the user. We also need rules to reset "ignores" on changes.

closed 2019-01-20 by davids5