PX4/PX4-Autopilot #10625: Upgrading through QGC seems to flash the wrong firmware.

Williangalvani, 2018-10-02:
Flashing new firmware on a PX4 FMU V2:

- Flashing PX4 **beta 1.7.4** reads **1.8.0** 
( QGCdownloads from [/**beta**/px4fmu-v2_default.px4](http://px4-travis.s3.amazonaws.com/Firmware/beta/px4fmu-v2_default.px4))

- Flashing PX4 **stable 1.8.0** reads **1.8.0dev** 
(QGC downloads from [/**stable**/px4fmu-v2_default.px4](http://px4-travis.s3.amazonaws.com/Firmware/stable/px4fmu-v2_default.px4))

- Flashing PX4 **master 1.8.0** reads **1.8.0dev** 
(QGC downloads from [/**master**/px4fmu-v2_default.px4](http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v2_default.px4))

By "reads" I mean what it shows on the summary page.

I'm building QGC from latest master.

dagar, 2018-10-03:
Thanks for reporting, I'll look at fixing this.

tubeme, 2018-10-08:
@Hey Daniel we have this problems almost at every new release. Is there any permanent solution?

tecdee, 2019-01-10:
The issue is still not solved. I still cannot load 1.8.2, as QGC still loads 1.8.1.

julianoes, 2019-07-11:
@tecdee can you check again?

julianoes, 2019-10-09:
No response, closing.

closed 2019-10-09 by julianoes