PX4/PX4-Autopilot #10638: UUID 18 open issues

davids5, 2018-10-03:
1) Eliminate the use of msg.uid  and PX4_CPU_UUID_WORD32_UNIQUE_x in code base OR at a minimum make sure they represent the slowest changing digits in the ID on all arch.

2) Legacy formats removed ? 
  Remove all DEPRECATED data types, #defines and functions and comments.
  board_get_uuid
  board_get_uuid32_formated
  board_get_mfguid_formated

3) Implication on Uavcan: HardwareVersion:unique_id holds 16 bytes. We used 12 in the past written to offset 0-11.

@pavel-kirienko - PX4_GUID is now 18 bytes. 

*  offset:0         1         2         -           17
  <ARCH MSD><ARCH LSD><MSD CPU UUID>...<LSD CPU UUID> padded to 16 bytes with 0s, if mfg id is < 16. 

I am assuming rather than change the UAVCAN lib to be 18 bytes we should use the most rapidly changing from (node to node) bits of the ID to fill the 16 bytes. In this case it is the last 16 bytes of the  PX4_GUID as the ID for HardwareVersion:unique_id in application?

Will this be a problem if that PX4 application use 16 bytes  and we do not rev the bootloader to use the 16 bytes?

pavel-kirienko, 2018-10-04:
A 16-byte (128-bit) UID is required for UAVCAN; it cannot be changed by modifying libuavcan alone because it is required by the protocol specification (affects wire compatibility). The same UID must be used both by the bootloader and by the application.

davids5, 2018-10-04:
Given Pavels input, and if we do not want to change the bootloaders, we have the option on the STM32 uavcan application truncate the PX4 UUID to the last 12 bytes with 0 fill in the 12-15 offsets. This will match the bootloader.

julianoes, 2019-07-11:
@davids5 is there still work to do here?

davids5, 2019-07-11:
@julianoes - yes but with UAVCAN 1.0 pending there will be a lot to do, including redoing all the bootloaders. 

@pavel-kirienko can we get a larger UUID in the 1.0 spec?

pavel-kirienko, 2019-07-11:
> can we get a larger UUID in the 1.0 spec?

The argument seems too PX4-specific. Rest of the world [uses 128-bit UUID](https://en.wikipedia.org/wiki/Universally_unique_identifier) so I would prefer to stay with the existing standard.