PX4/PX4-Autopilot #10639: DeltaQuad unable to recover from gust

DavidMackie, 2018-10-03:
After attempting to climb to 30 m in mission mode a moment is applied due to a gust. This causes the DeltaQuad to bank excessively sending it into a pitch down dive. We attempted to take over manual control but we didn't have enough altitude to recover. 

We flashed our flight controller (Dropix V2.1) with the Master branch (Updated 24th September). We then tested our DeltaQuad in manual mode quickly to ensure everything was working correctly and we set up a simple mission (with a few way-points then a land only in QC mode). It began climbing normally until 24 m when a gust caused the drone to enter a roll. The drone wasn't able to recover from this and caused an adverse roll followed by a nose down attitude until contact with the ground.

Link to the log file: https://review.px4.io/plot_app?log=992ecb6b-8410-4bd6-91e2-e82a7fa5e826

Our drone is a Skywalker X8 with VTOL modules.

sanderux, 2018-10-03:
Hi @DavidMackie 

The log you posted is not from a Vertical Technologies DeltaQuad as far as i can tell. Is this your own build?

DavidMackie, 2018-10-04:
This was flying with the master branch

Speterius, 2018-10-04:
My best theory is IMU failure between 1:15 and 1:16. Can anyone confirm this?

sanderux, 2018-10-04:
@DavidMackie can you please confirm if this was your own built vehicle or an official Vertical Technologies DeltaQuad?

If it was your own vehicle then i should note that the parameters associated with the airframe you selected have been specifically tuned for the proprietary hardware in the official DeltaQuad and should not be used on other vehicles. You would be better of using the Generic Quad Delta airframe as the settings in the Vertical Technologies DeltaQuad will likely not work on your vehicle.

From your log it seems like there was a motor failure (motor 4).

sanderux, 2019-01-20:
Addressed in 1.9

closed 2019-01-20 by sanderux