PX4/PX4-Autopilot #10655: MAGNETOMETER Inconsistent on Pixhawk 2.1 Cube and HERE GPS.

AlexisTM, 2018-10-05:
**Describe the bug**
Pixhawk 2.1 (Cube) and HERE GPS have inconsistent magnetometers.

The HERE GPS has a 180° rotation and the Pixhawk 2.1 has a 90° rotation.

I confirm that the HERE GPS is monted with the arrow aligned with the Pixhawk 2.1 Cube arrow. 

I performed the exact same test on an IRIS+ and the arrow was correctly pointing towards the physical location.

**To Reproduce for External Mag (HERE)**
Steps to reproduce the behavior:
1. Enable only the external mag (CAL_MAG0_EN = Enabled, Disabled MAG1, MAG2, MAG3)
2. Set the CAL_MAG_PRIME parameter to the CAL_MAG0_ID
2. Point the copter towards a physical direction (or north)
3. Check the computed north location
4. Observe an offset between 120 and 180° for MAG0.

**To Reproduce for Internal MAG (Cube)**
1. Enable only the MAG1, MAG2, MAG3
2. Set the CAL_MAG_PRIME parameter to the CAL_MAGX_ID
2. Point the copter towards a physical direction (or north)
3. Check the computed north location
4. Observe an offset around 90° for all internal MAGs.

**Log Files and Screenshots**

For the external mag (HERE GPS): The physical orientation is towards B&B Dr. Vee

[image]

For all internal mags (Cube): The physical orientation is towards B&B Dr. Vee

[image]

**Drone (please complete the following information):**
- Tarot Peeper with T10x gimbal
- Pixhawk 2.1 (Cube) 
- HERE GPS

[image]

dagar, 2018-10-05:
Can you verify which sensor is in your particular Here GPS? My understanding is that it's changed over time.

Could you run listener across all mags?

``` Bash
    listener sensor_mag -i 0
```
Repeat for instance (-i) 1, 2, 3.

AlexisTM, 2018-10-05:
The HERE GPS is the MAG0 and stayed MAG0, I checked it by disconnecting the HERE and it says `MAG0 timeout`

AlexisTM, 2018-10-05:
[image]

[image]

dagar, 2018-10-05:
This is an hmc5883 and correctly marked as external. 

[image]

This line is suspect - https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L166-L167

ealdaz, 2018-10-11:
We are experiencing issues with the headings reported by the internal magnetometers on the cube, so was interested by this article, however the sensor_mag command doesn't seem to work in the same way as per the posts.
We are using Pixhawk 2.1 and Firmware 1.8.0

The command seems to ignore the flags -i X
```
NuttShell (NSH)
nsh> listener sensor_mag -i 3
never published
nsh> listener sensor_mag -i 2
nsh> listener sensor_mag -i 1
nsh> listener sensor_mag -i 0
nsh> listener sensor_mag 

TOPIC: sensor_mag instance 0 #1
 sensor_mag_s
	timestamp: 113290950 (0.008811 seconds ago)
	error_count: 0
	x: 0.161
	y: 0.030
	z: 0.405
	range_ga: 48.000
	scaling: 0.002
	temperature: 60.105
	device_id: 263202
	x_raw: 19
	y_raw: -133
	z_raw: 475
	is_external: 0
```

AlexisTM, 2018-10-16:
Could it simply be a wrong driver due to the many versions available?

https://github.com/ArduPilot/ardupilot/issues/6633#issuecomment-316697574

```
In a standard first revision here, there is a HMC5983, and an LIS3MDL
In the Here + an additional MCU9250 as well as the others.

In future Here and here plus units, only a single ICM20948 is fitted due to both the LIS3MDL and the HMC5983 being EOL.

In the 2.0 cube... (by 3DR) there is a HMC5983 on SPI, and an LSM303D
In the 2.1 cube... (ProfiCNC/Hex) there are two MPU9250's, and the LSM303D
```

AlexisTM, 2018-10-16:
https://github.com/ArduPilot/ardupilot/issues/6633#issuecomment-349816217

Could be this: 10k between 5V and GND to solve a brownout issue on the compass of the HERE. 

Will not solve the 90° of the Cube2.1

AlexisTM, 2018-10-17:
> This is an hmc5883 and correctly marked as external.
> 
> [image]
> 
> This line is suspect - https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L166-L167

The Z magnetometer shows a negative value for the HERE magnetometer. That one should be positive

TobiMiller, 2018-11-14:
Same here with Pixhawk4 and GPS Module which is provided in pack. I have disabled internal mag.

i have got a yaw problem in manual flight, where it turns ccw : https://logs.px4.io/plot_app?log=374cafc0-6b92-4e57-87d2-0750f0ab03b6 
is it depending on this problem?

AlexisTM, 2018-11-15:
Related to: #10614

closed 2019-02-27 by stale[bot]

bigbellmercy, 2019-04-29:
> Could it simply be a wrong driver due to the many versions available?
> 
> [ArduPilot/ardupilot#6633 (comment)](https://github.com/ArduPilot/ardupilot/issues/6633#issuecomment-316697574)
> 
> ```
> In a standard first revision here, there is a HMC5983, and an LIS3MDL
> In the Here + an additional MCU9250 as well as the others.
> 
> In future Here and here plus units, only a single ICM20948 is fitted due to both the LIS3MDL and the HMC5983 being EOL.
> 
> In the 2.0 cube... (by 3DR) there is a HMC5983 on SPI, and an LSM303D
> In the 2.1 cube... (ProfiCNC/Hex) there are two MPU9250's, and the LSM303D
> ```
> 
> Lower, it also says:
> 
> ```
> Once the problem occurs, rebooting ardupilot (on the flight controller) does not resolve the problem, only powering down the flight controller resolves it.
> ```
> 
> I was indeed rebooting with max 2 seconds without power and using reboot command.

I leave a comment to just inform an additional info about the types of IMU in Pixhawk2.1 (Cube) because it is hard to find the exact info about its IMUs.

According to the info above (from the Ardupilot github) and my software check and analysis, the installed IMUs on Pixhawk2.1 (Cube) are as following:

1st IMU: MPU9250 for accel, gyro and mag
2nd IMU: L3GD20 for gyro and LSM303D for accel and mag
3rd IMU: MPU9250 for accel, gyro and mag

Particularly, there are two MPU9250.

> In the 2.1 cube... (ProfiCNC/Hex) there are two MPU9250's, and the LSM303D

Pixhawk2.1 H