PX4/PX4-Autopilot #10657: Need to prevent Arming when in configurations (e.g. modes) not approprate for takeoff.

Antiheavy, 2018-10-05:
### Issue:
We see a large number of users who forget to place a Takeoff waypoint at the start of their mission or do not properly select the Takeoff waypoint when arming.   This causes safety problems:
- Launch detection is bypassed
- fixed wing takeoff logic is bypassed
- motor/propeller may spool up at unexpected times based on LandDetection
- may cause crashes upon hand launch because the takeoff logic is ignored
- may fly to an unexpected waypoint (because obviously the user is already not paying attention)

Notes:
Our fixed wing only ever operates in Mission, Hold, and Return modes, manual modes are not used.
We depend on Mission Takeoff waypoint for proper takeoff performance. 
QGC "Start Mission" command switches the vehicle to Mission mode and then Arms the vehicle, but doesn't force which waypoint should be active (it probably should) or check for a takeoff waypoint (firmware probably should).

### Possible solutions:
I'm not sure the exact solution, but here are some ideas:

**Consider adding a new parameter "COM_ARM_MIS_TAKEOFF":**
- if enabled, at pre-arm checks it would require a Takeoff waypoint to be in the mission and active if Arming in Mission mode.
- Warning messages: "Cannot Arm, Mission Takeoff Waypoint missing", "Cannot Arm, Mission Takeoff Waypoint not active".
-  This gets tricky because QGC will sometimes add mission items prior to the first real waypoint which makes this complicated and may require some cleaver checking of the mission.

**Consider blocking arming from some mission modes:**
- Is arming ever appropriate if the vehicle is in Land, Hold, or Return modes?  Should we block arming in those modes? 

**Fix auto-disarm for hand launched vehicles which use launch detection:**
- currently auto-disarm cannot be used for some hand launched vehicles because LandDetector switches in and out of Landed state during ground handling and auto-disarm will disarm the vehicle if held still for too long before launch.
- this would add a layer of safety if the vehicle was armed in a manual mode and then switched into an Auto mode.
- related Issue and PR:
https://github.com/PX4/Firmware/issues/10076
https://github.com/PX4/Firmware/pull/9471 

**Related QGC PRs:**
These QGC PR's from @dagar might be part of a larger solution.
https://github.com/mavlink/qgroundcontrol/pull/5732
https://github.com/mavlink/qgroundcontrol/pull/5741

dagar, 2018-10-08:
Adding it as a feasibility check is pretty quick, but I'm wondering if we can get a simpler solution by focusing on state of the mission during the arming transition itself. This cleanly works around reversing potentially valid use cases where you might not want a mission to have to start with takeoff.

Antiheavy, 2018-10-08:
Adding it to feasibility check (enabled by Param) has some pros and cons.

- Pros: reminds user at time of Upload that they forgot a Takeoff WP, simple to implement
- Cons: doesn't check that the active WP at time of Arm is an appropriate one, don't necessarily need a Takeoff WP if the user uploads a new mission during flight

I think the Cons can be mitigated by:
1) Add a pre-arm check to see that the active mission item is appropriate (e.g. either the Takeoff WP is active, or at least the Takeoff WP the first real WP in the mission items list after the active mission item).
2) Just live with the fact that if you enable this optional Param, then ALL your missions must contain Takeoff waypoints - this is actually good from a consistency standpoint anyway.

Not sure the best param name in this case, either COM_ARM_MIS_TAKEOFF or MIS_TAKEOFF_REQ

Antiheavy, 2018-10-09:
QGC will sometimes put non location based items at the start of a mission.  Feasibility Checker may need to be smart enough to allow some MAV_CMD_'s in the list prior to the Takeoff Waypoint.

These are the commands I know about that QGC can put at the start of a mission, @DonLakeFlyer are there others?
178	MAV_CMD_DO_CHANGE_SPEED
530	MAV_CMD_SET_CAMERA_MODE
205	MAV_CMD_DO_MOUNT_CONTROL
206	MAV_CMD_DO_SET_CAM_TRIGG_DIST
2000 	MAV_CMD_IMAGE_START_CAPTURE

DonLakeFlyer, 2018-10-18:
I wouldn't base this on what QGC may or may not put prior to a takeoff command. I would base this on what commands are invalid prior to a takeoff command which would lead to a mission which doesn't make sense. Only the firmware really knows the answer to that and should be able to implement knowing nothing about what QGC does or does not do. It needs to classify mission commands internally somehow from an ordering standpoint. 

Example invalid sequence: WAYPOINT, TAKEOFF
Example valid sequence: start video (whatever command that is), TAKEOFF

I would contend the same problem with this:
Invalid sequence: ..., LAND, WAYPOINT
Valid sequence: ..., LAND, video off

QGC in reality does nothing to prevent the addition of any command before/after any other command. Only the firmware knows whether that may or may not be valid.

Antiheavy, 2019-01-28:
FYI @TSC21

Antiheavy, 2019-01-30:
If we can get Qgroundcontrol to switch the active waypoint to 1 when the "start mission" slider is used, that would alleviate the need for the pre-arm check to verify what waypoint is active.  This assumes that all people using the "start mission" slider in Qgroundcontrol actually intend to start the mission from the "start" (fair assumption I would say).  @DonLakeFlyer suggested some required changes to Qgroundcontrol that would be required to make this work here: https://github.com/mavlink/qgroundcontrol/pull/5741#issuecomment-341509796   This could be part of a solution.

Then the mission feasibility checker only has to check that the first "position type" item in the mission is a "Takeoff".

Antiheavy, 2019-02-26:
This related PR helps address part of this issue:  https://github.com/PX4/Firmware/pull/11531

The remaining parts of this issue could be addressed in a new state-machine architecture to prevent arming in inappropriate states.

Antiheavy, 2019-07-15:
we are working on a couple PRs in a custom branch now that further address this issue.  We might be able to bring to upstream.  FYI @kjkinney

Antiheavy, 2019-10-13:
We are working on some improvements for this that we will. E pushing upstream in the next month. We will keep the issue open for now. @khkinney