PX4/PX4-Autopilot #10659: FlightTask: implement MPC_LAND_ALT1/2

sanderux, 2018-10-07:
MPC_LAND_ALT1 and MPC_LAND_ALT2 should be used for a final approach speed during landing.
According to the param spec the vehicle should descend at MPC_Z_VEL_MAX_DN speed and slow down between MPC_LAND_ALT1 and MPC_LAND_ALT2 to MPC_LAND_SPEED.

There is a massive energy consumption benefit in having this

Currently this is not implemented in the FlightTask: https://github.com/PX4/Firmware/blob/bee6a6b8b08d30b36475628e0750edcb6038d798/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L117

The params are used for takeoff which does not seem to be documented behavior: https://github.com/PX4/Firmware/blob/bee6a6b8b08d30b36475628e0750edcb6038d798/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L128

Perhaps implementing it as indicated currently is not a good idea, the altitudes default to 10 and 5m above home and we might be flying over a different altitude.

There are a few options on how this can be implemented safely;
1. Within a small radius around home / for RTL only
2. Only when a range finder is installed
3. If QGC has given the terrain altitude
4. When enabled by a param
5. When enabled in the waypoint (explicitly chosen in QGC)

If there are any other ideas on how to do this please share.

Log demonstrating the behavior: https://review.px4.io/plot_app?log=642eae62-6203-4ab3-b372-ef0b58f01e4c
MPC_LAND_SPEED was 0.7
MPC_Z_VEL_MAX_DN was 2.5

/cc @dagar @bresch @LorenzMeier

raygamero, 2018-10-10:
I flew an extensive test on this using the lidar lite on 1.8 and found that this only works in position mode.  In position mode the aircraft slows its descent appropriately.  Land and RTL  modes just use MPC_LAND_SPEED.

MaEtUgR, 2019-01-16:
I just heared this issue get referenced by @raygamero in the dev call today. Thanks @dagar for assigning.

raygamero, 2019-01-16:
@MaEtUgR 
Recent test on a sim with version 1.9.0 Alpha.  RTL  and LAND descents do slow down as desired.
Set for MPC_LAND_ALT1 of 10m and MPC_LAND_ALT2 of 5m and a 3m/s descent rate and a .7m/s land speed.  
I don't know why I didn't have it working in my 1.8 system.  I'll retest.

MaEtUgR, 2019-01-16:
@raygamero

**History**
- I introduced the feature in https://github.com/PX4/Firmware/commit/72fb7a5062989dcebc7607ebab0e870b72584106 (21 Mar 2017)
- It was in at 1.8 release (19 Jun 2018) and should be working fine, flight tasks were off by default https://github.com/PX4/Firmware/blob/96443b3cf3e3adaec8b2ebb6c83a19a92d40a13f/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2437-L2441
- It possibly had a regression with https://github.com/PX4/Firmware/commit/e51e1bbe7414abf0cb80f1618ca9c46ba36e84ad (merged 20 Jul 2018) where flight tasks became default. I actually made the commit but @Stifael did more work on that particular part so I made him the author. The change was well worth it. It could be that this slipped into 1.8 (19 Jun 2018) I'd need to check.
- At some point it was added back again for missions in takeoff and landing by @Stifael. The slow down e.g. is here in the code: https://github.com/PX4/Firmware/blob/c319ea7189df9953bd2dcb65f606315d96ddb5b8/src/lib/FlightTasks/tasks/AutoLine/FlightTaskAutoLine.cpp#L210-L214

**Short term todo**
For manual position flight I'll readd it very soon too.

JFYI We were discussing today about making it one altitude parameter at which to slow down because depending on your algorithm its not useful to have two altitudes. See https://github.com/PX4/Firmware/issues/11209#issuecomment-454738626.

**Investigating**
The first implementation which is in the 1.8 release does **not** use the distance sensor and only the altitude above home https://github.com/PX4/Firmware/blob/96443b3cf3e3adaec8b2ebb6c83a19a92d40a13f/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2412
The new implementation now uses the distance sensor whenever it's available like can be seen here https://github.com/PX4/Firmware/blob/c319ea7189df9953bd2dcb65f606315d96ddb5b8/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L149-L162
Maybe we could even use the barometer for flying in altitude mode without GPS. I think we should have a unified way to provide the task or any other interested module with the best estimate of the current altitude above ground then it gets super easy.

bresch, 2019-03-25:
Fixed by: https://github.com/PX4/Firmware/pull/11540

closed 2019-03-25 by bresch