PX4/PX4-Autopilot #10660: Microhard 2.4 i2c interference with SF11 lidar

ryanjAA, 2018-10-07:
Put a new Microhard 2.4 link for telemetry on a test plane (a plane that's flown many many times with no issues and only component changed was the addition of the Microhard unit) and it caused chaos to the SF11 lidar. We isolated it to 2 potential issues. First thought was current to the Microhard link taken from the telem1 port so we lowered the power output from the microhard unit but that still caused interference (just removed low avionics power warning on the pixhawk). Then finally moved the unit another 10cm forward and away from the lidar (this particular plane had the lidar right next to the microhard module, not in the component bay which is 60cm away) and the interference went away.

So, a few takeaways, the Microhard units need to probably be about 10cm away from most i2c units (suppose we should just put that in the docs somewhere, like how the initial lidar lite units caused gps interference), second would be that they can not be run at full power from telem1 port, will cause low avionics warnings. Lastly, and this is probably a bigger overall issue, we need to probably think about lidar health and treating it as a sensor. Reasoning being that if we are using a fixed wing and are relying on lidar for landing, etc, its importance is greatly increased and it will still pass preflight checks since interference didnt cause it to fail, only to just be intermittent. We amended our preflight checks on the lidar later that day but that is pretty extensive on this one sensor when the initial preflight process regarding lidar data has always been sufficient. Any thoughts on reporting health of this sensor during flight? Maybe via QGC? Too hard to simply monitor it as a stand alone but I think it is probably a good idea since if the data gets bad and you end up coming in for an auto landing and the previous X second are filled with junk data, the rest is history....

Running stable and FMUv2/3

Antiheavy, 2018-10-08:
Which microhard are you using?  Is it the radio itself or the antennas that you think might be causing EMI?

ryanjAA, 2018-10-08:
Pico 2400 (p2400). Definitely not the antenna, it was externally mounted (wing type from Persistent Systems) and even when turning the power down on the Microhard, it still did it until moving the unit away from the lidar.

closed 2019-02-04 by stale[bot]