PX4/PX4-Autopilot #10669: Landing pattern for RTL and LAND in VTOL UAVs

moreba1, 2018-10-09:
Hello ,
In the vtol drones, when RTL start at high altitudes, the drone reaches the top of the home at high altitude and the Battery consumption is too much in order to discend in the MC mode.
I have an professional idea for RTL. This idea needs to change in autopilot code and the Qgroundcontrol software . we can design a landing pattern such as a fixed wing landing pattern using the software. The UAV must returns to a loiter(altitude) waypoint in FW mode ,Then when it’s reach to direction of home, it must at that altitude start the transition to to MC and reach to the top of the home. Then reduce the altitude to Land .
In this landing pattern we can define this parameters:
1- distance(to Land point) and heading of loiter waypoint
2- loiter altitude
3- landing waypoint

tubeme, 2019-01-21:
I think it shouldn't be dependent on QGC because it is part of the critical procedures, ie should be self contained in the core of PX4.

julianoes, 2019-07-15:
FYI @RomanBapst

closed 2020-09-26 by moreba1

dagar, 2020-09-26:
Was this resolved?

moreba1, 2020-09-26:
> 
> 
> Was this resolved?

Yes, in the daily build version.