PX4/PX4-Autopilot #10672: Should UAVCAN Motor parameters be in the documentation?

hamishwillee, 2018-10-10:
The XML output of our generated parameter doc injects this file of [UAVCAN Motor Parameters](https://github.com/PX4/Firmware/blob/master/src/lib/parameters/parameters_injected.xml) (which presumably goes to QGC).

This is not included in the documented Full parameter reference. Is this a bug? (ie should I add this?)

dagar, 2018-10-10:
The entire thing is worth reviewing in detail. Last I tried the uavcan parameter sync wasn't working correctly anymore. FYI @mhkabir 

We also have a full UAVCAN setup headed to the test flight team.

mhkabir, 2018-10-11:
We currently don't have a proper maintainer for UAVCAN. @pavel-kirienko is your best choice :)

pavel-kirienko, 2018-10-11:
I am probably the least wrong person to ask about that, but I am pretty far from being a maintainer of anything in PX4.

dagar, 2018-10-11:
@mhkabir @pavel-kirienko perhaps we could do a coordinated pass to ensure all UAVCAN functionality in PX4 is working as intended. From that point I'll volunteer to keep it working with help from @PX4/testflights and CI.

pavel-kirienko, 2018-10-11:
@dagar this is much appreciated. You might be aware that we are currently drafting up UAVCAN v1.0 (finally), which is going to fix one glaring design mistake in UAVCAN v0 at the cost of offering somewhat limited backward compatibility. Please follow our progress here: https://forum.uavcan.org/c/dev

Seeelefant, 2018-10-12:
That`s strongly appreciated 🥇 .  Currently we have more than one UAVCAN compatibility problem within PX4, whereas the CAN bus is honestly made for such environments like UAVs, much more reliable thatI2C

hamishwillee, 2018-10-14:
@pavel-kirienko @dagar @mhkabir So the way I read @dagar 's response was we need your help to get this to good state which we can then maintain. To keep this moving along, when is a reasonable time for you guys to work together on this co-ordinated pass?

hamishwillee, 2019-01-30:
?

closed 2019-02-13 by stale[bot]