PX4/PX4-Autopilot #10677: Dodecarotor cox bottom rotors don't spin

GregGoldman, 2018-10-10:
Hi all,

I am trying to use the dodecarotor cox configuration on a pixhawk 2 using QGroundControl. Unfortunately, none of the bottom rotors spin when connected to the aux ports. I tried probing the pins with an oscilloscope: the main pins do just fine, but the aux pins get no signal. Is this a known issue, and has anyone used this build successfully?

rallegab, 2018-12-04:
I think this is the same issue as mentioned by @1nvoker (https://github.com/PX4/Firmware/issues/9719). I'm having the same issue so I will second both of these. 

I know the code for this was written by @imcnanie (https://github.com/PX4/Firmware/pull/7532/files/fa93fe96096559d8a88eb46b1f3e55645d0fd6e6) back in June 2017, so it might be that one of the updates since then on how the mixing is done in PX4 (now it uses the px_generate_mixers.py script with .toml geometry files instead of using the multi_tables.py script) has screwed something up, and noone did a flight test on the new implementation before commiting.

I've been looking at it myself but I only know a limited amount about the PX4 code, so I've been unable to find a solution so far. Did you make any progress on this problem yourself? It would also be great if anyone that knows more about this might be able to help.

imcnanie, 2019-03-04:
@GregGoldman @rallegab Just saw this. I'll see if I can get it running in a couple of days. Bumping to keep this from closing.

clanegge, 2019-03-13:
@GregGoldman @rallegab @imcnanie Anybody has any updates on this issue?

rallegab, 2019-03-14:
@clanegge Sorry, I've had no time to look at this since, and frankly I'm on other projects at the moment. I might move back to this soon however, maybe in a month or so.

simounours, 2019-06-05:
Hello ! Any update about to make the dodecarotor works with PX4 ?

LorenzMeier, 2019-06-05:
@MaEtUgR Could you look into this? Thanks!

Mario2410, 2019-09-10:
Helo, also have the same problem here. Is there any update right now?
Thanks

MaEtUgR, 2019-10-21:
ok, I think the biggest problem I face to debug this is to reproduce it.
I'll try to have a test setup and pull in additional help if I don't find the problem.

Mario2410, 2019-10-24:
Hi Matthias,

thanks a lot for your answer.

I hope you can fix the problem.

Best regards

Mario

tiloro, 2019-12-04:
Any updates or info about which version is compatible with the configuration? Really having trouble here..

saumilss2, 2020-03-03:
I'm having exactly the same issue with Bottom Motors or DodecaRotor not spinning.
I'm using PixHawk 2.0 with px4_fmu-v3.

I have already tried extensive troubleshooting on this.
My takeaways:
 - In PX4_SITL mode, with 16 pwm available on /dev/pwm_output0, I can easily load 
   dodeca_top_cox.main.mix as well as dodeca_bottom_cox.aux.mix using following commands.
   
    mixer load /dev/pwm_output0 /dir/dodeca_top_cox.main.mix
    mixer append /dev/pwm_output0 /dir/dodeca_bottom_cox.aux.mix

- But on NuttX Target, there are 8 main pwm on PX4IO (/dev/pwm_output0) while 6 aux pwm on 
  PX4FMU (/dev/pwm_outptu1), so the "mixer append" doesn't really work.
  
  mixer load /dev/pwm_output0 /etc/mixers/dodeca_top_cox.main.mix
  mixer load /dev/pwm_output1 /etc/mixers/dodeca_bottom_cox.aux.mix

- Above should work from the NSH console, but I'm unable to load the bottom mixer on 
  /dev/pwm_output1

- However I tried "mixer load /dev/pwm_output1 /etc/mixers/quad_x.main.mix" as well as "mixer 
  load /dev/pwm_output0 /etc/mixers/dodeca_bottom_cox.aux.mix" and both of the commands 
  worked

- With that being said, it doesn't seem like there's anything incorrect with mixer files and other 
  scripts who process the mixer geometries. It's also fairly obvious that, loading mixer on first 4 pwm 
  aux outputs works, but px4fmu driver, definitely seems to have some issue communicating with 
  last 2 pwm aux outputs.

- I have been using 1.9.0 version of PX4 Firmware, I tried to play with increasing buffer lengths, 
  increasing PX4IO numbers etc, I also used the latest version of the firmware and the issue still 
 persists.

- This is really frustrating to get a very expensive drone flying. I really appreciate if developers can 
  look into it once more. I have provided as much information as I could. If I'm missing anything, 
  please feel free to reply to my comment.

MaEtUgR, 2020-03-12:
Sorry that this issue went missing in the cracks. To be honest I don't consider myself IO communication or mixer configuration expert and don't have a Dodecarotor setup but I have a pixhawk 4 and can run some tests on there. Thanks a lot for all the investigations that narrow down the problem to be loading mixer configuration `/etc/mixers/dodeca_bottom_cox.aux.mix` on the IO pwm device `/dev/pwm_output1`!

saumilss2, 2020-03-13:
Hey Matthias,
Thank you so much for looking more into this.
**/etc/mixers/dodeca_bottom_cox.aux.mix** mixer file loads just fine on 
PX4IO pwm device, **/dev/pwm_output0**, but it is a struggle to load the same mixer file on
PX4FMU aux pwm device, **/dev/pwm_output1**
Very bizarre.

spyvel, 2020-03-31:
Hello everyone.

I am also trying to get a dodecarotor running with px4. The bottom rotors won't spin. I am new to all this but during my tries I noticed a couple of things which seem weird and may help someone who is more experienced with the firmware code.

In QGC Mavlink console, I typed the command:

pwm info -d /dev/pwm_output1

the response was:

device: /dev/pwm_output1
WARN  [fmu] warning: trim values not valid - no mixer loaded

and after that it shows info about the channels and channel groups. Then typing the following command:

param show PWM_AUX_TRIM1

the response was:

Symbols: x = used, + = saved, * = unsaved

 735/1298 parameters used.

It seems that this parameter is empty (for every channel, 1 - 8). It doesn't even have the default value. Searching the parameters catalogue in QGC, I noticed that this parameter is not shown although the rest of the PWM_AUX parameters along with the PWM_MAIN parameters are shown properly (even the PWM_MAIN_TRIM parameters). Thus PWM_AUX_TRIM can't be set through the parameters menu. It can be set from the Mavlink console but it wouldn't make any difference because after a reboot pixhawk will see again these parameters as empty. I also tried to load the mixer after I have set the PWM_AUX_TRIM parameters equal to 0.0:

mixer load /dev/pwm_output1 /etc/mixers/dodeca_bottom_cox.aux.mix

The response was:

ERROR [mixer] failed to load mixers from /etc/mixers/dodeca_bottom_cox.aux.mix
ERROR [mixer] failed to load mixer

Trying to do the same to output0, is executed successfully. So, maybe something goes wrong regarding the way the PWM_AUX_TRIM parameters are treated when someone chooses the dodecarotor frame. If another frame is chosen, these parameters are visible in the QGC parameters menu.

I hope this will contribute to the solution of the problem.

MaEtUgR, 2020-04-01:
@spyvel That analysis is very useful. From the dev call today I have the impression that this issue is related to #14116 . Unfortunately I don't have the debugger in my quarantine to check but there might be a way to get the necessary output to the FMU and print it.

MaEtUgR, 2020-04-15:
It's definitely a RAM issue on the IO chip. There are some optimizations done in https://github.com/PX4/Firmware/pull/14129 and upcoming but I think it was mentioned the largest mixer still doesn't load 😬

spyvel, 2020-05-13:
@LorenzMeier
@MaEtUgR

Hello everyone.

It's been almost 1.5 months since my post in this topic and I can see that there is some progress. This matter is of great importance for my project and it's really a pity that the dodecarotor configuration is included in the QGC but it can't work properly.

However, I want to ask if the cause of the problem has been found and if we can expect a solution soon. Is it possible that this solution will be included in the upcoming firmware version? Is there anything we can do in QGC (e.g. edit some settings) to bypass this problem or we have to wait for the new firmware?

P.S. In my previous post I forgot to mention that in the dodecarotor configuration, whenever I tried to start the motors, the green PWR indicator of the FMU was turned off as soon as the motors started to move (of course only the I/O PWM controlled motors were moving). When the motors were stopped the indicator was turned back on.

spyvel, 2020-07-25:
Hi!

Any progress regarding the dodecarotor? I just tried PX4 v. 1.11.0-rc1 but with no luck. Will it work with the upcoming stable firmware version?

bys1123, 2020-09-07:
#14129 Fixed my problem.

dagar, 2020-09-07:
> #14129 Fixed my problem.

Thanks for the followup.

closed 2020-09-07 by dagar