PX4/PX4-Autopilot #10678: CI Failures due to Clang Tidy in unrelated files to active PR

mcsauder, 2018-10-11:
**Describe the bug**
Hi, I'm having trouble with active PRs due to Clang Tidy failures in files not touched by my active PR.  The root cause of the failures is not entirely obvious, and this occurrence has appeared to affect multiple PR's in the past few weeks. (And... Yes, I will also do some digging into fixing this bug that I am filing. ;) )

**To Reproduce**
Submit a PR.  Wait for CI tools to complete.

**Expected behavior**
One would guess that only files impacted by the PR would be those able to cause issues.  This appears to not be the case, however it is likely due to CI tool improvements while the code body remains deficient. 

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
[image]

**Additional context**
Reference PR #10629

fredowski, 2018-10-12:
The same for PR #10414 .

dagar, 2018-10-15:
@fredowski that particular one was a real failure that slipped into master.

@lamping7 this might be a case where it's acceptable to retry once after a failure.

lamping7, 2018-10-29:
I've watched this loosely and haven't noticed a failure here. I'm going to close for now. Feel free to re-open if it comes back up or if there is a better solution.

lamping7, 2018-10-29:
I take that back. I just found one. http://ci.px4.io:8080/blue/organizations/jenkins/PX4%2FFirmware/detail/PR-10774/1/pipeline
https://github.com/PX4/Firmware/pull/10774

@dagar ideas?