PX4/PX4-Autopilot #10679: Position setpoint drift during takeoff

sanderux, 2018-10-11:
A strange update in the local position setpoint seems to occur during takeoff on a fairly recent branch (few weeks old):

https://review.px4.io/plot_app?log=1d33a68b-c88b-420c-8e8f-4a4382675963

During a mission, The vehicle takes off 4 meters in the air then suddenly changes position. as seen on local position setpoint X/Y. it does resume its takeoff to final altitude.

The position it moved to seems like it is the location where the first home lock occurred. 

[image]

@dagar @RomanBapst @bresch

sanderux, 2018-10-11:
Possible cause:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L670-L697

@Stifael

sanderux, 2018-10-11:
@Stifael our branch did not yet have this one:
https://github.com/PX4/Firmware/blame/55b9f76a162d10976ad2ec79a4efa250b81a03cb/src/modules/mc_pos_control/mc_pos_control_main.cpp#L695-L696

Could that be the cause?

sanderux, 2018-10-11:
MIS_TAKEOFF_ALT was set to 15. The setpoint takeoff was 60m
Here's another log where i cherry-picked your flight-task reactivate, it indeed did not help.
https://logs.px4.io/plot_app?log=ebb2fcdc-418e-44cb-b42e-8f60340595c8

sanderux, 2018-10-11:
I tried disabling the vx and vy modifications during takeoff, that seems to make it smoother, but at takeoff it still has a little offset it seems, but no more sudden jerk.
https://logs.px4.io/plot_app?log=bb6cc36f-2797-43c7-9fd7-cfe18ef8cb41

dagar, 2018-10-15:
The first log (https://review.px4.io/plot_app?log=1d33a68b-c88b-420c-8e8f-4a4382675963) appears to be the result of initiating mission and then moving the vehicle.

When the mission is first evaluated a TAKEOFF mission item is created at the current location. This is poor navigator behaviour that can be fixed.

RomanBapst, 2018-10-16:
@sanderux Can you explain exactly how to reproduce this behavior? I tried various things in SITL but cannot reproduce a large offset like this.
As @dagar  said the takeoff position setpoint is created [here](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L626) as soon as the takeoff mode is activated. So even after a reposition it should use the current position and not the one prior to the reposition.

closed 2019-02-04 by stale[bot]