PX4/PX4-Autopilot #10686: Consistantly failing preflight checks

baumanta, 2018-10-15:
**Describe the bug**
In SITL ROS with gazebo: After launching the simulation, the drone can not be armed. The error satements say: "Primary accelerometer not found" if I try to arm, the I get "Arm/Disarm temporarily rejected"

This happens for any Firmware version after this [commit](https://github.com/PX4/Firmware/commit/c7e572d2c2bb21ae3dfdbff637022946d9f9b72c)

**To Reproduce**
Steps to reproduce the behavior:
1. Launch PX4 node
2. Launch Mavros node
3. Launch Gazebo
4. Spawn drone model
5. Try to arm over QGC

I used the following launch file from inside the local_planner [package](https://github.com/PX4/avoidance).
`<launch>
    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>
    <arg name="ns" default="/"/>
    <arg name="model" default="iris_depth_camera"/>
    <arg name="world_path" default="$(find local_planner)/../sim/worlds/simple_obstacle.world" />
    <arg name="build" default="posix_sitl_default"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
    
    <arg name="gcs_url" default="" />   
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />
    <arg name="mavros_transformation" default="-1.57" />

    <param name="use_sim_time" value="true" />

    
    <include file="$(find px4)/launch/px4.launch">
    </include>

    
    <group ns="$(arg ns)">
        <include file="$(find mavros)/launch/node.launch">
            <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
            
            <arg name="config_yaml" value="$(find local_planner)/resource/px4_config.yaml" />
            <arg name="fcu_url" value="$(arg fcu_url)" />
            <arg name="gcs_url" value="$(arg gcs_url)" />
            <arg name="tgt_system" value="$(arg tgt_system)" />
            <arg name="tgt_component" value="$(arg tgt_component)" />
        </include>
    </group>

    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world_path)" />
    </include>

    
    <node name="spawn_model" pkg="gazebo_ros" type="spawn_model" output="screen"
          args="-sdf -database $(arg model) -model iris">
    </node>
</launch>`

**Expected behavior**
I would expect the drone to arm and be operational

mrivi, 2018-10-15:
@dagar @baumanta I saw this as well. The problem is that the parameters are not correctly loaded and `CAL_*_PRIME` is always 0.

dagar, 2018-10-15:
Let's get something in the CI system representative of this usage.
I thought you would have hit the default parameters here https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L98, but apparently not.

mrivi, 2018-10-30:
@dagar @baumanta -  I did a bit of debugging with @julianoes . Here's what we figured out

On a clean build of `make posix_sitl_default gazebo` [AUTOCNF](https://github.com/PX4/Firmware/blob/c7e572d2c2bb21ae3dfdbff637022946d9f9b72c/ROMFS/px4fmu_common/init.d-posix/rcS#L90) is true and the parameters `CAL*_PRIME` are set. Those parameters are saved into `~/Firmware/build/posix_sitl_default/tmp/rootfs/eeprom/parameters`.

Now that the Firmware is built, we start the avoidance. The launch script is in the initial comment of the issue. `AUTOCNF` is false because it isn't a new build and the parameters are retrieved from the saved `paramters` file. However when starting PX4 from ROS the used file is not `~/Firmware/build/posix_sitl_default/tmp/rootfs/eeprom/parameters` but `~/.ros/eeprom/parameters` (which still contains the parameters from the old Fimware). 

To fix the issue you either need to remove the `parameter` file in `~/.ros` or run `catkin purge` to wipe the `~/.ros` folder. `catkin clean` is not enough since it only effects the build folder of the catkin workspace.

mrivi, 2019-04-09:
This is still a issue. If the vehicle type is changed from ros, the new parameters specific to that vehicle are not loaded correctly. Deleting `~/.ros/eeprom/parameters` make the new parameters load

julianoes, 2019-06-04:
Not stale, this just cost me 1h!

mrivi, 2019-06-04:
@dagar this is a quite annoying issue for anyone using ROS. I try to fix it without success. Could you look into this?

julianoes, 2020-05-25:
@baumanta is still an issue?

baumanta, 2020-05-25:
@julianoes I'm actually not sure. I have not used SITL in months

mhkabir, 2020-05-25:
Yeah this is still an issue

dagar, 2020-05-25:
> Yeah this is still an issue

What are you seeing exactly and how often?

mhkabir, 2020-06-04:
From running gazebo SITL on master:

```
ERROR [sensors] fewer accels than calibrations, resetting all CAL_GYROx_ID
```
After this, sim loading just freezed, presumably because it was waiting for lockstep which never happened.

Remove `~/.ros` fixed it.

dagar, 2020-09-05:
Finally resolved in master?

dagar, 2020-09-07:
Please reopen if this is still an issue.

closed 2020-09-07 by dagar