PX4/PX4-Autopilot #10694: posix_sitl_default gazebo error with OpenCV 4

tomcattigerkkk, 2018-10-16:
when I pull Firmware through git and make the project posix_sitl_default gazebo, there occurs an error. Has anyone met this error before?

`ninja: Entering directory `/home/arnold/Develop/Firmware/build/posix_sitl_default'
[3/7] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.14
-- Using C++17 compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/arnold/Develop/Firmware/build/posix_sitl_default/build_gazebo
[4/7] Performing build step for 'sitl_gazebo'
FAILED: cd /home/arnold/Develop/Firmware/build/posix_sitl_default/build_gazebo && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/arnold/Develop/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/15] Linking CXX shared library libgazebo_uuv_plugin.so
[2/15] Linking CXX shared library libgazebo_lidar_plugin.so
[3/15] Linking CXX shared library libgazebo_imu_plugin.so
[4/15] Linking CXX shared library libgazebo_vision_plugin.so
[5/15] Linking CXX shared library libgazebo_multirotor_base_plugin.so
[6/15] Linking CXX shared library libgazebo_motor_model.so
[7/15] Building CXX object OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o
FAILED: /usr/bin/c++    -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -I/home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -isystem /usr/local/include -isystem /usr/local/include/opencv -fpic -O3 -DNDEBUG -MMD -MT OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o -MF OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o.d -o OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o -c /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp
In file included from /usr/local/include/opencv2/core.hpp:52:0,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/local/include/opencv2/core/cvdef.h:440:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
 #  error "OpenCV 4.x+ requires enabled C++11 support"
    ^
In file included from /usr/include/c++/5/array:35:0,
                 from /usr/local/include/opencv2/core/cvdef.h:445,
                 from /usr/local/include/opencv2/core.hpp:52,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/local/include/opencv2/core/base.hpp:58:0,
                 from /usr/local/include/opencv2/core.hpp:54,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/local/include/opencv2/core/cvstd.hpp:389:12: error: expected ‘,’ or ‘...’ before ‘&&’ token
     Ptr(Ptr&& o);
            ^
/usr/local/include/opencv2/core/cvstd.hpp:389:16: error: invalid constructor; you probably meant ‘cv::Ptr<T> (const cv::Ptr<T>&)’
     Ptr(Ptr&& o);
                ^
/usr/local/include/opencv2/core/cvstd.hpp:390:25: error: expected ‘,’ or ‘...’ before ‘&&’ token
     Ptr& operator = (Ptr&& o);
                         ^
In file included from /usr/local/include/opencv2/core/cvstd.hpp:1033:0,
                 from /usr/local/include/opencv2/core/base.hpp:58,
                 from /usr/local/include/opencv2/core.hpp:54,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/local/include/opencv2/core/ptr.inl.hpp:256:16: error: expected ‘,’ or ‘...’ before ‘&&’ token
 Ptr<T>::Ptr(Ptr&& o) : owner(o.owner), stored(o.stored)
                ^
/usr/local/include/opencv2/core/ptr.inl.hpp:256:1: error: prototype for ‘cv::Ptr<T>::Ptr(cv::Ptr<T>)’ does not match any in class ‘cv::Ptr<T>’
 Ptr<T>::Ptr(Ptr&& o) : owner(o.owner), stored(o.stored)
 ^
[... 1367 lines cut ...]
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/include/c++/5/bits/stl_algobase.h: In instantiation of ‘static _BI2 std::__copy_move_backward<false, false, std::random_access_iterator_tag>::__copy_move_b(_BI1, _BI1, _BI2) [with _BI1 = cv::Point_<float>*; _BI2 = cv::Point_<float>*]’:
/usr/include/c++/5/bits/stl_algobase.h:604:58:   required from ‘_BI2 std::__copy_move_backward_a(_BI1, _BI1, _BI2) [with bool _IsMove = false; _BI1 = cv::Point_<float>*; _BI2 = cv::Point_<float>*]’
/usr/include/c++/5/bits/stl_algobase.h:614:5:   required from ‘_BI2 std::__copy_move_backward_a2(_BI1, _BI1, _BI2) [with bool _IsMove = false; _BI1 = cv::Point_<float>*; _BI2 = cv::Point_<float>*]’
/usr/include/c++/5/bits/stl_algobase.h:649:8:   required from ‘_BI2 std::copy_backward(_BI1, _BI1, _BI2) [with _BI1 = cv::Point_<float>*; _BI2 = cv::Point_<float>*]’
/usr/include/c++/5/bits/vector.tcc:339:4:   required from ‘void std::vector<_Tp, _Alloc>::_M_insert_aux(std::vector<_Tp, _Alloc>::iterator, const _Tp&) [with _Tp = cv::Point_<float>; _Alloc = std::allocator<cv::Point_<float> >; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<cv::Point_<float>*, std::vector<cv::Point_<float> > >; typename std::_Vector_base<_Tp, _Alloc>::pointer = cv::Point_<float>*]’
/usr/include/c++/5/bits/stl_vector.h:925:17:   required from ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = cv::Point_<float>; _Alloc = std::allocator<cv::Point_<float> >; std::vector<_Tp, _Alloc>::value_type = cv::Point_<float>]’
/home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:377:51:   required from here
/usr/include/c++/5/bits/stl_algobase.h:549:18: error: ambiguous overload for ‘operator=’ (operand types are ‘cv::Point_<float>’ and ‘cv::Point_<float>’)
      *--__result = *--__last;
                  ^
In file included from /usr/local/include/opencv2/core.hpp:58:0,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/arnold/Develop/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:
/usr/local/include/opencv2/core/types.hpp:170:13: note: candidate: cv::Point_<_Tp>& cv::Point_<_Tp>::operator=(const cv::Point_<_Tp>&) [with _Tp = float]
     Point_& operator = (const Point_& pt);
             ^
/usr/local/include/opencv2/core/types.hpp:171:13: note: candidate: cv::Point_<_Tp>& cv::Point_<_Tp>::operator=(cv::Point_<_Tp>) [with _Tp = float; cv::Point_<_Tp> = cv::Point_<float>]
     Point_& operator = (Point_&& pt) CV_NOEXCEPT;
             ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:171: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1`

TSC21, 2018-10-16:
What version of OpenCV you have "really" installed on your computer? Cause I see calls to OpenCV 2 on the build process...

tomcattigerkkk, 2018-10-17:
hi, @TSC21 
I've solved this problem by adding 
`set (CMAKE_CXX_STANDARD 11)`
in the file 'Firmware/Tools/sitl_gazebo/external/OpticalFlow/CMakeLists.txt' to make sure the c++ standard be 11.
I'm not sure if this is an individual problem or a bug. 

> What version of OpenCV you have "really" installed on your computer? Cause I see calls to OpenCV 2 on the build process...

TSC21, 2018-10-17:
It's a corner case. This doesn't happen in the current setup we support.

closed 2018-10-17 by TSC21

nikhilmarda, 2022-03-04:
i see a similar issue, but i don't know where to add the set command as well. i am running a test .cpp code to see if the setup I have is good or not