PX4/PX4-Autopilot #10695: Mocap message data lost for few seconds and fail to estimate local position (LPE)

andrespalomino, 2018-10-16:
Hi all,
I'm having some issues of missing data when publishing on /mavros/mocap/pose. It suddenly happens in some tests and the local position estimation fails due to those small gaps on the mocap message. 
I'm using 921600 bauds for the companion link and mocap data is streamed at 20hz from the onboard pc to pixhawk. 

I'm also making sure the points of the mocap arrives consistently to the onboard pc, so I believe the issue could be in the mavros publication or the prosessing on the pixhawk. Other published topics like the /mavros/setpoint_raw/local don't seem to be corrupted

The log of a single point at 0,0,2 is below and also a screenshot of the end of offboard mode where the drone should be stable but the missing points appear and the local estimation fails.

[https://review.px4.io/plot_app?log=05c12597-ebda-4e70-aec6-b83029438932](url)

[image]

TSC21, 2018-10-16:
LPE suffered some major changes since latest PX4 stable release. I advise you to try upstream master instead and come back to us with results.

andrespalomino, 2018-10-17:
I believe the LPE could improve by using the master but the missing mocap/pose data could still be present? Do you have any thoughts why this could be happening?

TSC21, 2018-10-17:
I would check the link between the onboard computer and the PX4. Also, which baud rate are you using?

andrespalomino, 2018-10-25:
I found some communication gaps on the UDP message coming from the mocap message but I changed to EKF2 and now the local estimation works better.

Now, I've a question about the yaw estimation on EKF2 as it seems to diverge sometimes from the vision yaw and EKF Yaw error happens afterwards.  I needed to reboot the system or it takes several minutes to converge to the vision yaw estimate. 
Do you have any hints for this or could you redirect me to some same issues and answers? Should I open a new question for this?

I set
EK2_AID_MASK = 24
EKF2_HGT_MODE = Vision
EKF2_EV_NOISE = 0.10

jgoppert, 2018-11-13:
Closing as stale, please repost your ekf2 question if required as a new issue.

closed 2018-11-13 by jgoppert