PX4/PX4-Autopilot #10706: Improve the module doc parser - subcategories/empty defaults

hamishwillee, 2018-10-17:
Propose the following improvements to the [docs API/tooling](https://github.com/PX4/Firmware/blob/v1.8.0/src/platforms/px4_module.h) for the [module documentation](https://dev.px4.io/en/middleware/modules_main.html)

1. Support for sub-categories. 
   - This is intended for categories like rangefinders - would be hard to find among all the other drivers.
   - This might be defined like:
      ```
      PRINT_MODULE_USAGE_NAME("sf1xx", "driver");  // module name, category
      PRINT_MODULE_USAGE_SUBCATEGORY("distance_sensor")   //subcategory
      ```
    - subcategories would be defined in own files named *category_subcategory.md*
    - subcategories would be auto-linked from near top of category page
    - subcategory page title would be something like 
       ```
       # subcategory (category)
       ```
    - If we like the idea of subcategories but not the work, alternative is to just use existing category mechanism for each subcategory. This works, but would not be possible to do automatic linking and have parent category in title.
1. Support for default to be declared "not applicable" and not rendered
   - Currently on many APIs you have to declare a default value and this always gets rendered.
      ```
      void PRINT_MODULE_USAGE_PARAM_INT(char option_char, int default_val, int min_val, int max_val,
		const char *description, bool is_optional);
      ```
      We should be able to specify either some sort of nothing pointer or -1 to mean "not applicable". 
   - If the 'not applicable" value is set for `default_val` then the text and value for default value should not be rendered in the markdown
   - Ideally this same mechanism should also be allowed for max, min and other possibly not applicable values (though these are not rendered)

hamishwillee, 2018-10-17:
@bkueng I've assigned this to you as best person to work on the script. As discussed in https://github.com/PX4/Firmware/pull/10676#issuecomment-430471579

When/if you do this I can go back and fix up the other example (or you could do it as part of testing?)

bkueng, 2018-10-17:
I can look into it (though I can't promise it will be soon).

hamishwillee, 2018-10-17:
Thanks. No urgency!

hamishwillee, 2019-01-30:
I'm still interested in this.

PR #11353 (2019-02-08): Improve the module doc parser - subcategories/empty defaults
Fixes #10706

@hamishwillee this adds support for empty defaults and subcategories (we currently have module docs only for one distance driver). The subcategories need to be added to the TOC of the dev-guide as well.
closed 2019-02-08 by bkueng