PX4/PX4-Autopilot #10718: Build Failed: "Recipe for Target 'px4fmu-v2_default' failed"

mujahid-abdulrahim, 2018-10-18:
I have been trying to build the px4 firmware for the px4fmu-v2, but I continue to get an error message:

"recipe for target px4fmu-v2_default failed"

There are numerous threads/bug reports on this issue, and I've tried to implement the recommended solution.  Initially, it seemed as though the problem was that I was using gcc 4.2 and the build was a few bytes too large for the 1MB limit on Pixhawk 1.  I installed gcc 7, but it looks like I didn't do it correctly.

Here is the complete result of the build attempt:

22:39 $ make px4fmu-v2_default
ninja: Entering directory `/home/mujahid/git/uf_px4_firmware/build/nuttx_px4fmu-v2_default'
[1/637] git submodule platforms/nuttx/NuttX/apps
[2/637] Building px4io-v2
ninja: Entering directory `/home/mujahid/git/uf_px4_firmware/build/nuttx_px4io-v2_default'
[1/196] Generating nuttx/include/nuttx...ersion.h, nuttx/include/nuttx/config.h
FAILED: cd /home/mujahid/git/uf_px4_firmware/build/nuttx_px4io-v2_default/NuttX/nuttx && make --quiet --no-print-directory context >nuttx_build.log
make[3]: gcc: Command not found
make[3]: *** [mkconfig] Error 127
make[2]: *** [tools/mkconfig] Error 2
ninja: build stopped: subcommand failed.
Makefile:170: recipe for target 'nuttx_px4io-v2_default' failed
make[1]: *** [nuttx_px4io-v2_default] Error 1
[4/637] Copying NuttX/apps to /home/mu...re/build/nuttx_px4fmu-v2_default/NuttX
FAILED: cd /home/mujahid/git/uf_px4_firmware && make --no-print-directory nuttx_px4io-v2_default
[4/637] Copying NuttX/nuttx to /home/m...re/build/nuttx_px4fmu-v2_default/NuttX
ninja: build stopped: subcommand failed.
Makefile:176: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

"gcc not found" seems like an ominous warning, and typing "gcc -v" reveals the error "The program 'gcc' is currently not installed".

Any suggestions for how I can repair gcc so I can get back to building px4?

hamishwillee, 2018-10-18:
Try reinstall by following [these instructions](http://dev.px4.io/en/setup/dev_env_linux_ubuntu.html#nuttx-based-hardware)?

If you have never had PX4 builds working on this machine then I would use the [build scripts here](http://dev.px4.io/en/setup/dev_env_linux.html)

mujahid-abdulrahim, 2018-10-18:
Thanks for your suggestion.  I tried reinstalling it per those suggestions, but I received the same error.

I had previously built PX4 on this machine, and the misconfiguration resulted from trying to update GCC to resolve the 'recipe for target failed' error.

hamishwillee, 2018-10-18:
I'm sorry, not sure what to suggest then. Between builds though you should do a `make distclean` to really really clean up build artifacts. The scripts definitely work on a clean build of Ubuntu 16.04.

closed 2019-02-04 by stale[bot]