PX4/PX4-Autopilot #10720: Option to use on-board RGB LED for status reporting during temp cal

CarlOlsson, 2018-10-18:
**Describe problem solved by the proposed feature**
In the online temperature calibration the status is indicated by an external RGB LED. However, if I want to calibrate a large number of Pixracers at the time it becomes cumbersome to connect anything to the boards.

**Describe your preferred solution**
Add an option to use the on-board LED instead

**Additional context**
I want to add this feature but I would need some guidance on where to start

CarlOlsson, 2018-10-31:
this is actually very straight forward, just use ioctl to control the led as is done in commander.

CarlOlsson, 2018-10-31:
> Would you be interested in upstreaming this? :) This is super valuable for large-scale batch calibrations!

It depends on the amount of time it would take, for our application it is completely fine to just use ioctl in send_events.

Do you know why the current architecture don't control the on-board LED in exactly the same way as an external? (using the led_control topic)

mhkabir, 2018-10-31:
@davis5 @dagar @simonegu can probably answer that best.

davids5, 2018-10-31:
I have a mental model that had been solidified on FMUv5. There are status LEDs and UI LED's the status LEDs are the ones used by the bootloader, at boot and by commander to indicate status (see https://github.com/PX4/Firmware/pull/10579). The number and color varies.  The UI LED are the I2C or PWM user facing RGB LEDs. This one is used to inform the user of state and is consistent in that is always had RGB. 

If you point me to the calibration code I can make some suggestions on how to work with the code when 10579 comes in.

CarlOlsson, 2018-11-01:
> If you point me to the calibration code I can make some suggestions on how to work with the code when 10579 comes in.

@davids5 [here](https://github.com/PX4/Firmware/blob/master/src/modules/events/temperature_calibration/task.cpp#L188) is the control of the external LEDs in the temperature calibration

davids5, 2018-11-02:
A factory pattern could be used in the calibration code. An interface would be retrieved from the factory that supports the simple LED or the Orb based RGB LED, via an API. But the available color choices (per board) and blink patterns will be different  and the Least common set (LCS) of indication possible would have to be used.

Assuming we chose to extend the uOrB led_control message to include control for the status LEDs (not preferred) OR better yet create a status LED message, you have the same set of issues.

The first issue is that the thread of execution that times the animation of the LEDs lives in each of the lower level drivers (rgbled, rgbled_pwm). See the the code that call `led_controller.update(led_control_data)`. This logic would need to be added to the simple LED driver. The up side here is that the to loop counting code in commander could be replaced with code that uses the Simple LED interface. 

The second issue is that logic to arbitrate the usage of the LEDs would be needed. Commander is normally using the status led and now the calibration code would need to steal them. 

Even a parallel usage of the macros from #10579 added inline would have to deal with the LCS, threading and arbitration issues.

CarlOlsson, 2018-11-06:
thanks David for the in depth explanation. This is out of scope for my current work at the moment but I will reopen the issue such that the discussion is easier to find if someone else would have the resources to work on this improved architecture.

closed 2019-02-18 by stale[bot]