PX4/PX4-Autopilot #10731: vtol crash during normal flight

JohnSnowball, 2018-10-21:
Plane lost radio during mission, then we waited until we thought the plane should be back, afterwards we drove along the mission waypoints, then we found plane crashed, upside down on the ground with **fixed-wing pusher still running**!!!!   I checked log, found no error shown, however, log stopped while plane is flying.

here is the log link:
https://logs.px4.io/plot_app?log=152bee65-95b5-4211-adcd-4c4051c1674c

Standard vtol with 4 rotors and 2 pushers. Made at least 10 flights, everything was normal except for this time.....

LorenzMeier, 2018-10-21:
Can't really comment on it as you're flying non-standard software. Where is the source code for it?

JohnSnowball, 2018-10-21:
@LorenzMeier  Code is derived from master v1.6, I added a RTK gps driver and nothing more...

LorenzMeier, 2018-10-21:
the last v1.6 release was well over a year ago plus adding an RTK driver can indeed cause software faults, so I'm closing this. We, unfortunately, can't offer free support on the community level for old releases and customized software. Feel free to reach out to one of the companies offering commercial support: http://px4.io/commercial-support/

closed 2018-10-21 by LorenzMeier