PX4/PX4-Autopilot #10732: MC land detector question

garfieldG, 2018-10-21:
Hi,

After looking at the land detector code I have a questions :
1. Why do we use vz in https://github.com/PX4/Firmware/blob/26e85736a52f5dd6cf2395c06a151d17ae0512f1/src/modules/land_detector/MulticopterLandDetector.cpp#L193, but we use derived z in https://github.com/PX4/Firmware/blob/26e85736a52f5dd6cf2395c06a151d17ae0512f1/src/modules/land_detector/MulticopterLandDetector.cpp#L200? shouldn't we use derived in both cases?

Thanks in advanced:)

RomanBapst, 2018-10-27:
@garfieldG I'm not too familiar with that code but I think it has to do with the fact that the estimated velocity can have a considerable bias when landing which makes land detection harder. The derivative of the height state might suffer less and I think that's why it's used.

garfieldG, 2018-10-31:
@RomanBapst Thank for response!
But I still can't understand why if we landed less than 2 seconds ago we use vz, otherwise we use vertical derivative. Shouldn't we use the same one (estimated vertical velocity **or**  vertical derivative) in both cases?
@Stifael maybe you know how to answer on this one? relates to https://github.com/PX4/Firmware/commit/26e85736a52f5dd6cf2395c06a151d17ae0512f1

Stifael, 2018-10-31:
if I remember it correctly, that commit entered the firmware against my advice. That commit was part of a test branch to check if it would improve the detection, which I could not verify (maybe it does, I was just not able to do verify it). The fact that the derivative term is not always used is probably because of inconsistency. It probably would make sense to evaluate once more the difference between z-deriv and vz.

PR #11172 (2019-01-26): Slow landing improvement suggestion
**Context**
Search for "landing" after this comment: https://github.com/PX4/Firmware/pull/9731#issuecomment-403600121
hopefully fixes #9696
fixes #10732
hopefully fixes #10001

**Test data / coverage**
SITL still works. Release testing for the upcoming H520 update shows a massive speedup in land detection.

**Describe problem solved by the proposed pull request**
When introducing the flight tasks architecture almost the whole legacy position controller was replaced a lot of things were modularized and simplified from a developers perspective. One of the thin [... cut]
closed 2019-01-26 by MaEtUgR