PX4/PX4-Autopilot #10734: FlightTask disable velocity control not working

RomanBapst, 2018-10-22:
**Describe the bug**
In the FlightTask.hpp header there it the following [statement](https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/FlightTask/FlightTask.hpp#L187):

> Setpoints that are set to NAN are not controlled

This is not true anymore if the current velocity of the vehicle exceeds the maximum velocity of the vehicle.
In that case the velocity setpoint will be constrained to the maximum allowed velocity [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L221) which results in the velocity setpoint and the current velocity not producing a zero velocity error anymore.

I noticed this issue because I've been trying to implement a transition flighttask which just controls height during transition but ignores x/y states. However, due to the above described bug this is not possible.

**Expected behavior**
Any state that is set to NAN in a flighttask should strictly not be controlled otherwise the interface is useless.

**Possible solutions**
- modify the maximum velocity parameter during a transition
- pass a control structure to the controller with which we can explicitly disable / enables state to be controlled

@Stifael @MaEtUgR Any other suggestions?

Stifael, 2018-10-22:
There is an interface-mapping [https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L100] in the PositionControl file who should take care of it. 
What probably is mission from 
>Setpoints that are set to NAN are not controlled

is that one set of sepoint needs to be valid [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L110-L169) (either position, velocity or thrust). If that is not the case, then it will enter failsafe.

>I noticed this issue because I've been trying to implement a transition flighttask which just controls height during transition but ignores x/y states.

For that you would need to set thrust[0]=[thrust[1]=0 and position and velocity along xy to NAN.

Stifael, 2018-10-22:
Did you not receive a failsafe message?

RomanBapst, 2018-10-22:
@Stifael Thanks for the description!

> Did you not receive a failsafe message?
It's definitely not going into failsafe mode.

> For that you would need to set thrust[0]=[thrust[1]=0 and position and velocity along xy to NAN.
I'm exactly doing that! This is the code from my flighttask:

```
thrust_setpoint(0) = _thrust_setpoint(1) = 0.0f;
_thrust_setpoint(2) = NAN;
_position_setpoint *= NAN;
_velocity_setpoint *= NAN;
_position_setpoint(2) = _transition_altitude;
```

I 'm really sure the problem is right [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L221) when the maximum velocity of the vehicle is exceeded.

Stifael, 2018-10-22:
Your implementation seems to be correct and is the same as used in altitude, that is why I am a bit skeptical why it shouldn't work.

Let's go though PositionControl file with the inputs from above:
- xy: pos_sp(0/1)=pos(0/1) [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L151) and vel_sp(0/1)=vel(0/1) [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L158)
- z: _pos_sp(2)=_transition_altitude and thr_sp(2)=0 and vel_sp(2)=0 [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L113-L115)
- [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L216): vel_sp_position(0/1)=0 since _pos_sp(0/1)=_pos(0/1)
z: will be non-zero 
- [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L217): vel_sp(0/1)=0 because vel_sp_position(0/1)=0 and vel_sp(0/1)=0 from the interface mapping
- [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L221) the function constrainXY will have inputs: (0,0), (0,0), speed_xy.
- in [constrainXY](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/Utility/ControlMath.cpp#L111-L115), with zero inputs the output should just be 0. 

Given the above explanation, I don't know yet where the bug is. Maybe you can add some printfs. Hence, before changing anything in the implementation structure, I suggest to first figure out what causes the bug. In principle you are doing the exact same thing that happens during altitude

Stifael, 2018-10-22:
Ok, now I understand it what you mean.
Yep, the logic breaks when the velocity is larger than the max.

RomanBapst, 2018-10-22:
@Stifael Hm, are you sure this will work? If you set the setpoint to zero any non-zero velocity will create a velocity error.
Looking at the code I'm actually confused about something else now
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L280

What if I demand a zero thrust setpoint in xy direction? Wouldn't in that case the position controller ignore it and run the velocity controller instead?

Stifael, 2018-10-22:
> if you set the setpoint to zero any non-zero velocity will create a velocity error.

I meant to set the velocity and velocity-setpoint to 0 as it is done [here] (https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L161) when there is no velocity estimation present. The same can be done with position and position setpoint.

About: 
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/PositionControl.cpp#L280

The point follows the logic above. If there is a valid thrust setpoint that you want to execute, you will set position setpoint and velocity setpoint such that they have no effect. There are two ways to do that:
1. set position-sp and velocity-sp eqaul to position and velocity respectively 
2. set position-sp, velocity-sp, position and velocity to 0

I chose the first solution because I did not want to adjust the position and velocity. On the other hand, setting the states and the setpoints to 0 might be the safer choice, and in fact you just showed that 1. does not always hold.

RomanBapst, 2018-11-06:
Fixed in #10745

closed 2018-11-06 by RomanBapst