PX4/PX4-Autopilot #10743: TFMini LIDAR not working on v5 board

nikan8, 2018-10-23:
Hello everyone,
please advise on the following issue
When plugged into any of the serial ports on the Pixhawk4 (Holybro) board, TFMini Lidar's readings fail to show up on QGC (the Distance_Sensor variable is not present), even though the parameter SENS_EN_TFMINI is set to '1' (Enable). The issue disappears when the same Lidar gets connected to FMU v2 board running PX4.

To reproduce:
1. take a newly flashed Pixhawk 4 board with PX4 1.8 stable
2. Plug TFMini LIDAR to Serial 4 port (UART &I2CB)
3. Connect the board to USB
4. Launch QGC, set SENS_EN_TFMINI to '1' (Enable) and reboot the board.
5. Open analyze widget, the distance sensor variable will not be present.

If exactly the same procedure is repeated with newly flashed Pixhawk v 2 board, the issue disappears.
I believe, this is a driver issue.
Thank you in advance.

mhkabir, 2018-10-23:
The driver is not started ln the correct port for  FMUv5 in v1.8

In master, this has changed and now serial ports are configured differently. https://docs.px4.io/en/peripherals/serial_configuration.html

nikan8, 2018-10-23:
Apologize for a lot of newbie asking, where can I find 'master' release? thank you.

mhkabir, 2018-10-23:
Flash the Developer Build through Qgroundcontrol

nikan8, 2018-10-23:
thank you. much appreciated.

nikan8, 2018-10-23:
Just another quick question to clarify, if I flash developer build, will I be able to configure SENS_TFMINI_CFG and SER_TEL4_BAUD? Thank you.

nikan8, 2018-10-23:
mhkabir, could you please advise where in the code (stable build) serial port assignment is made for RFMINI rangefinder?

nikan8, 2018-10-23:
the issue is resolved. thank you

closed 2018-10-23 by nikan8

mhkabir, 2018-10-23:
@nikan8 What were the changes necessary? Can you share it soon that we fix it/backport it?

nikan8, 2018-10-23:
I just set Sys_companion=0 so that TELEM 2 port is free of MAVLInk, and connected TFMINI to TELEM2, as the code seemed to indicate pls see [(https://github.com/PX4/Firmware/blob/stable/ROMFS/px4fmu_common/init.d/rc.sensors)]

nikan8, 2018-10-23:
Ideally, it may be better to have a parameter like sens_en_tfmini in stable build to be able to choose a port of one's own liking. But anyway, thanks for your support.