PX4/PX4-Autopilot #10747: ERROR:bootloader reports invalid operation

FrancescoDAlterio, 2018-10-23:
I am trying to upload the version of the PX4 flight controller on Intel-aero ready to fly drone.

I have followed those guides:

 

https://dev.px4.io/en/setup/building_px4.html

https://docs.px4.io/en/flight_controller/intel_aero.html

 

but anytime I try to run the command: make aerofc-v1_default upload, the sistem erases the old flight controller but while installing the new one (the PROGRAMING phase) it stops at 0% saying:

ERROR: bootloader reports INVALID OPERATION

hamishwillee, 2018-10-23:
@zehortigoza This something you could help with?

zehortigoza, 2018-10-23:
I will try to look at this but in the mean time you can copy the firmware to Aero RTF board by hand(I don't remember where the firmware is build now) and then use Aero script to flash it: aerofc-update.sh <firmware>

closed 2019-01-22 by mhkabir