PX4/PX4-Autopilot #10751: Firmware compilation issue

Ankur1014, 2018-10-24:
**Describe the bug**
The compilation of PX4 Firmware is failing with a different syndrome every time. Every time, I use the command 'make px4fmu-v5_default' to build, I encounter a new issue. I have listed some of the syndromes below.

**Syndrome 1:**
/mnt/c/px4/Firmware/build/nuttx_px4fmu-v2_default/NuttX/nuttx/include/arch/irq.h:53:10: fatal error: arch/chip/irq.h: No such file or directory

**Syndrome 2:**
/home/batserve/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include/sys/_types.h:168:5: error: 'wint_t' does not name a type; did you mean 'ino_t'?
     wint_t __wch;
     ^~~~~~
     ino_t

**Syndrome 3:**
/mnt/c/px4/Firmware/build/nuttx_px4fmu-v2_default/NuttX/nuttx/include/arch/armv7-m/irq.h:64:12: fatal error: arch/chip/chip.h: No such file or directory
In file included from /mnt/c/px4/Firmware/src/drivers/drv_hrt.h:44:0,
                 from /mnt/c/px4/Firmware/src/drivers/vmount/input_mavlink.cpp:46:
/mnt/c/px4/Firmware/build/nuttx_px4fmu-v2_default/NuttX/nuttx/include/inttypes.h:52:10: fatal error: arch/inttypes.h: No such file or directory
 #include <arch/inttypes.h>
          ^~~~~~~~~~~~~~~~~

**Syndrome 4:**
No rule to make target 'NuttX/nuttx/arch/arm/src/chip/chip/stm32f33xxx_adc.h', needed by 'src/systemcmds/hardfault_log/CMakeFiles/systemcmds__hardfault_log.dir/hardfault_log.c.obj'.  Stop.
CMakeFiles/Makefile2:4316: recipe for target 'src/systemcmds/hardfault_log/CMakeFiles/systemcmds__hardfault_log.dir/all' failed
make[3]: *** No rule to make target 'NuttX/nuttx/arch/arm/src/chip/chip/stm32f10xxx_rcc.h', needed by 'src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.obj'.  Stop.
CMakeFiles/Makefile2:4197: recipe for target 'src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/all' failed
make[2]: *** [src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/all] Error 2
[ 80%] Built target drivers__teraranger
make[2]: *** Waiting for unfinished jobs....

**Syndrome 5:**
/mnt/c/px4/Firmware/build/nuttx_px4fmu-v2_default/NuttX/nuttx/arch/arm/src/chip/stm32.h:83:10: fatal error: stm32_lowputc.h: No such file or directory
 #include "stm32_lowputc.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.

After about 7 retries, I was able to complete the compilation. But still, once every 5 times, I get an error saying 'No rule to make target'

**To Reproduce**
OS: Windows 10
Install Windows Subsystem for Linux
git clone https://www.github.com/PX4/Firmware
git checkout v1.8.0

make px4fmu-v5_default

**Expected behavior**
The compilation should go through without issue

mhkabir, 2018-10-24:
We do not support building on WSL. Cygwin is supported : https://dev.px4.io/en/setup/dev_env_windows.html

closed 2018-10-24 by mhkabir

hamishwillee, 2018-10-25:
As @mhkabir says, Cygwin is the only supported Windows platform. 

That said, we do have instructions that have worked previously: https://dev.px4.io/en/setup/dev_env_windows_bash_on_win.html

So if you have set up as per those instructions then it should work (if it doesn't, post a bug against that page in the Devguide (there is icon at top of page for creating bug reports). 

But still recommend Cygwin!