PX4/PX4-Autopilot #10752: HIL no position lock

kuiskas, 2018-10-24:
**Describe the bug**
When I try HIL I cannot use any flight mode that implies GPS, altough QGC shown a 3d lock.  
Always getting a ERROR rejecting "flightmode" no position lock yet.Please retry.
**To Reproduce**
Steps to reproduce the behavior:
1. Start HIL
2. Try for example `commander takeoff` on the mavlink console.

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- using Pixhawk cube 2.1, Taranis Frsky & X8R receiver

**Additional context**
Using QGC v3.4.4, ubuntu 16.06 docker container and latest stable v 1.8.0 of PX4

bkueng, 2018-10-25:
Can you try with master? There was a fix here: https://github.com/PX4/Firmware/pull/10586.

bkueng, 2018-10-25:
Did you flash the fmu-v2 target? If so, some modules are missing there due to limited flash space. You can update the bootloader (https://docs.px4.io/en/advanced_config/bootloader_update.html) and flash fmu-v3, then it should work.

KeyneZzz, 2018-11-07:
> Thanks.
> I just tried and this problem is gone, but I did find another problem.
> When I am using my Pixhawk cube 2.1 I can arm but it does not takeoff in Gazebo (neither with the RC nor with the QGC mav_cmd).
> Today arrived the pixhawk 4 and I tried HIL and in this Hardware is working without any problems.
> Do you have any idea? I reset every parameter, got a fresh pull from PX4 but not finding the problem.
> Should I post this better in the other post and close this one?

i am also facing the problem cannot taking off with HITL in airsim. i tried a version in about mid sept, it worked normally . but with a version lastweek(nov 1st) it can just arm but never takeoff. there may be some problems during the period.

kuiskas, 2018-11-07:
Hi KeyneZzz, 
I was facing yesterday the same issue. I cannot fly in HIL, but it does arm. I hope someone in the community or dev. team has a clue and can help as. I will keep trying. If I find something I share the info

closed 2019-02-19 by stale[bot]