PX4/PX4-Autopilot #10757: DANGEROUS: VEHICLE_CMD_DO_REPOSITION broken on master

dakejahl, 2018-10-25:
Had a Drone fly into a pole and smash into the ground when issuing a VEHICLE_CMD_DO_REPOSITION. I looked at the log and everything looked fine. This is broken on current `master` 294af5daada2fa08da8d9f1c4310779683fe9750
https://review.px4.io/plot_app?log=cc6bccfc-44a5-4eca-bc2f-7a4be13a0a7a
[image]

As you can see we were just adjusting the altitude setpoint and nothing else.

I then tested this in SITL using `make posix_sitl_default gazebo` and issued via QGroundControl a **`goto`** command and saw the vehicle fly to the new location, but then fly super far an a random direction for a while, then returns back to that location, wanders around a bit more, flys off in the opposite direction... then I got bored of watching.
[image]

Isn't CI supposed to catch these types of bugs? This is extremely concerning to me.

dagar, 2018-10-25:
Do you know when it last worked?

> Isn't CI supposed to catch these types of bugs?

Yes, but it's mainly testing core controller and mission functionality so far. Every feature like this we care about needs targeted testing to prevent these kind of regressions.

bkueng, 2018-10-25:
> This is extremely concerning to me.

Agreed.

Fix is here: https://github.com/PX4/Firmware/pull/10760

closed 2018-10-25 by LorenzMeier

dakejahl, 2018-10-25:
Awesome, thanks for the quick fix.

PR #10760 (2018-10-25): fix reposition: set acceptance radius
Previously the acceptance radius was 0, so the FlightTaskAutoLine was
randomly changing yaw and sometimes going into a random direction (see  #10757).

There is still something else wrong in there, but this patch avoids the reposition bug. I see 2 problems:
- The FlightTaskAuto contains states and resets certain attributes when switching (https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/Auto/FlightTaskAuto.cpp#L468). In case of the reposition, the switches were random due to numerical inaccuracies. If possible we should avoid these sta [... cut]