PX4/PX4-Autopilot #10759: make px4fmu-v5_default menuconfig - does not work

RossosHEX, 2018-10-25:
**Describe the bug**
Board configuration step in porting guide https://dev.px4.io/en/hardware/porting_guide.html
make px4fmu-v5_default menuconfig
does not work

**To Reproduce**
Steps to reproduce the behavior:
$: ./ubuntu_sim_nuttx.sh (from https://dev.px4.io/en/setup/dev_env_linux.html)
$:~/src/Firmware$ make px4fmu-v5_default menuconfig

`$:~/src/Firmware$ make px4fmu-v5_default menuconfig
-- PX4 VERSION: v1.8.0-1581-g3e9e551
-- CONFIG: nuttx_px4fmu-v5_default
-- Build Type: MinSizeRel
-- The ASM compiler identification is GNU
-- Found assembler: /home/rossos/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-603-g10a0fef
-- Building and including px4io-v2
-- Using C++11
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- NuttX: px4fmu-v5 nsh cortex-m7
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default
ninja: Entering directory `/home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default'
[1/6] git submodule platforms/nuttx/NuttX/nuttx
[2/6] git submodule platforms/nuttx/NuttX/apps
[6/6] Running NuttX make menuconfig for px4fmu-v5 with nsh
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/nuttx/configs/px4fmu-v5/include
LN: include/arch/chip to arch/arm/include/stm32f7
LN: arch/arm/src/board to /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/nuttx/configs/px4fmu-v5/src
LN: arch/arm/src/chip to arch/arm/src/stm32f7
LN: platform/board to /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/platform/dummy
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/examples
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/gpsutils
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/system
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/netutils
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/interpreters
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/graphics
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/fsutils
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/canutils
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps/wireless
mkkconfig in /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/apps
/bin/sh: 1: kconfig-mconf: not found
Makefile.unix:504: recipe for target 'do_menuconfig' failed
make[1]: *** [do_menuconfig] Error 127
FAILED: cd /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/nuttx && make --no-print-directory --silent -C /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/nuttx CONFIG_ARCH_BOARD=px4fmu-v5 menuconfig && cp /home/rossos/src/Firmware/build/nuttx_px4fmu-v5_default/NuttX/nuttx/.config /home/rossos/src/Firmware/platforms/nuttx/nuttx-configs/px4fmu-v5/nsh/defconfig && cd /home/rossos/src/Firmware && platforms/nuttx/NuttX/tools/nuttx_defconf_tool.sh /home/rossos/src/Firmware/platforms/nuttx/nuttx-configs/px4fmu-v5/nsh/defconfig
ninja: build stopped: subcommand failed.
Makefile:176: recipe for target 'px4fmu-v5_default' failed
make: *** [px4fmu-v5_default] Error 1`

dagar, 2018-10-25:
Hi @RossosHEX 

You'll need to build and install one of the kconfig tools from NuttX tools. https://bitbucket.org/nuttx/tools/src/master/

Personally I like qconfig, but you need to get the qt dependencies installed.

[image]

Alternatively you can use the px4-dev-nuttx docker container which has kconfig-mconf preinstalled. https://hub.docker.com/r/px4io/px4-dev-nuttx/

RossosHEX, 2018-10-25:
Hello @dagar ,

Thank you for your reply.

Following your suggestions I did the following:

```
git clone https://bitbucket.org/nuttx/tools.git
cd tools/kconfig-frontends
sudo apt install gperf
./configure --enable-mconf --disable-nconf --disable-gconf --enable-qconf
make
sudo make install

```
It compiled and seemingly installed under /usr/local/lib/
However, neither *menuconfig*, nor *qconfig* see it.
Is it supposed to be under different directory?

[image]
[image]

yl5006, 2018-10-30:
you need install kconfig-frontends in  /usr
./configure --prefix=/usr

RossosHEX, 2018-10-30:
@yl5006 Thank you!
Your suggestion solved it. I am now able to get into menuconfig and qconfig

```
cd tools/kconfig-frontends
make clean
./configure --enable-mconf --disable-nconf --disable-gconf --enable-qconf --prefix=/usr
make
sudo make install
```

hamishwillee, 2018-10-30:
@RossosHEX Do you think you could please amend http://dev.px4.io/en/hardware/porting_guide.html#nuttx-menuconfig appropriately?

You can directly edit the page as a PR in github by clicking the link up the top of the page:
[image]

RossosHEX, 2018-10-31:
@hamishwillee I have done as you suggested.
Not very familiar with GitHub, but I hope my request went through.

RossosHEX, 2018-11-01:
Did it show up now? https://github.com/PX4/Devguide/pull/641

hamishwillee, 2018-11-01:
Thanks very much - yes it did!

yl5006, 2018-11-22:
@lukegluke use qconfig does not have this problem

lukegluke, 2018-11-22:
@yl5006 thanks for the feedback, but qconfig does have this problem

```
git clone https://github.com/PX4/Firmware.git

git clone https://bitbucket.org/nuttx/tools.git
cd tools/kconfig-frontends
sudo apt install gperf
./configure --enable-mconf --disable-nconf --disable-gconf --enable-qconf --prefix=/usr
make
sudo make install

cd Firmware
make px4-stm32f4discovery_default qconfig
```

and then git diff shows that
```
CONFIG_ARCH_BOARD_PX4_STM32F4DISCOVERY
CONFIG_ARCH_BOARD_PX4_STM32F4DISCOVERY
CONFIG_BOARD_CRASHDUMP
CONFIG_BOARD_RESET_ON_CRASH
CONFIG_BOARD_CRASHDUMP
CONFIG_BOARD_RESET_ON_CRASH
CONFIG_BOARD_HAS_PROBES
CONFIG_BOARD_USE_PROBES
```
were cleared

and make px4-stm32f4discovery_default
failed with nuutx_build.log:
```
No .version file found, creating one
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/osboxes/Firmware/build/nuttx_px4-stm32f4discovery_default/NuttX/nuttx//include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/osboxes/Firmware/build/nuttx_px4-stm32f4discovery_default/NuttX/nuttx//src
No directory at /home/osboxes/Firmware/build/nuttx_px4-stm32f4discovery_default/NuttX/nuttx//src
Makefile.unix:311: recipe for target 'arch/arm/src/board' failed
```

without qconfig everything is fine

closed 2019-03-06 by stale[bot]