PX4/PX4-Autopilot #10763: What is the buffer for LPE when switching between estimates in Pixhawk

siddhu45, 2018-10-26:
I am trying to implement a failsafe in an indoor environment using Pixhawk. I am trying some new things on the drone and want to use motion capture as my own failsafe. whenever the drone is going unstable or the behaviour is not as required, I give a keyboard input which is taken as an action to initiate failsafe. The drone then lands safely using the motion capture system. I am testing a different position feedback system but I am publishing both, actual and Optitrack [only during failsafe] on the same topic `/mavros/vision_pose/pose`. I am switching between them as needed.
My issue is as soon as I switch from my position estimates to mocap postion estimates, the drone takes some time to use the new data. and the failsafe works but not as intended. I want to know how can I get around this buffer and make pixhawk switch between two estimates almost instantaneously or atleast as fast as possible.

thanks for any help!

siddhu45, 2018-10-26:
Yes, I have basic understanding and know that instantaneous switch in estimates is not possible.
But what I meant was is there any other work around for this? Thanks!

jgoppert, 2018-11-13:
There is no buffer as discussed, but there is code to handle timeout and initialization. You can looks at the code for that here: https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/sensors/mocap.cpp#L9. Notice by default it takes 20 valid reads before it will start fusing. In general we recommend you move to EKF2 as LPE is nolonger supported.

closed 2018-11-13 by jgoppert

siddhu45, 2018-11-18:
Thanks, @jgoppert for the reference. Yes, we will try to switch to EKF2 soon but is not viable at the moment. I will look into the info you shared. I have a Q though for the implementation.
As I mentioned in the issue, I am publishing my original feedback on the vision topic and then sending the mocap data on the vision topic as well. Are you suggesting that when I want to switch the feedback, I publish on the mocap topic instead of vision topic? will this work? thanks and appreciate your response.