PX4/PX4-Autopilot #10771: LPE setup problem

YuYangguang, 2018-10-27:
I want to use the vicon system as the external position estimator. Therefore, I follow the guide [](http://dev.px4.io/en/ros/external_position_estimation.html). However,in the latest vesion of the Firmware, I couldn't find the corresponding parameters, such as ATT_EXT_HDG_M. I use the command 
"make px4fmu-v2_lpe" to build the code. Is there anything wrong?

jgoppert, 2018-11-13:
ATT_EXT_HDG_M is defined by attitude_estimator_q. Make sure that att_est_q it is being built. In general we are nolonger supporting LPE and recommend you move to EKF2.

closed 2018-11-13 by jgoppert

vkotaru, 2018-11-13:
So, does EKF2 support mocap_pose_estimate?

Thanks

hamishwillee, 2018-11-14:
@vkotaru 

> So, does EKF2 support mocap_pose_estimate?

Looks like it: http://dev.px4.io/en/tutorials/motion-capture-vicon-optitrack.html#ekf2