PX4/PX4-Autopilot #10778: Pixhawk 4 with RTK HERE + GPS

Haeryangkim, 2018-10-29:
Hi,
I'm planning to use Pixhawk 4 with RTK here + GPS

But it seems HERE+ GPS support Pixhawk 2 directly (support Pixhawk 3 by changing cable).

I can't find any information about Pixhawk 4 with related to RTK GPS.
Is there any tutorial for this?

Thanks.

Haeryangkim, 2018-10-30:
@Jaeyoung-Lim  I checked the document including the pinout map of pixhawk, I'm just wondering what the Korean pixhawk official seller says the here + RTK is for pixhwak 2 (I assume that is just cable problem). Therefore, I need to double-check it before I buy the expensive stuff.

Jaeyoung-Lim, 2018-11-01:
@Haeryangkim I wasn't aware that there was a 'official' seeler for pixhawk in Korea.

Anyway, the hardware version(pixhawk2= fmuv3) is what matters more than the actual numbering on the name. Pixhawk 4 is fmuv5.

The firmware supports RTK, so it should also work with pixhawk4.

proficnc, 2018-11-02:
There are two official resellers in S.Korea for the Cube and Here products. (Formally Pixhawk 2) 
The cube is FMUv2 and FMUv>5 depending on colour. 
However, the Here is compatible with all Cube and PX4 related hardware. Just may need to set up a cable depending on what hardware you have.
Plug and play with the cube of all versions

closed 2019-02-18 by stale[bot]

rmtorre, 2020-04-29:
Hello guys

I have the same issue. I have a Pixhawk 4 and a Here+ RTK. Does anyone knows if is just a pinout problem? Any updates, please?

Thanks

Phil4771, 2020-05-09:
Hi everybody, same problem for me
Anybody contacted holybro?
Any update?

Phil4771, 2020-05-09:
[image]

i Will try to do this

Phil4771, 2020-05-09:
Pixhawk 4 UART & 12C B port has the same pinout than pixhawk 3 pro
[Pixhawk4-Pinouts.pdf](https://github.com/PX4/Firmware/files/4603041/Pixhawk4-Pinouts.pdf)