PX4/PX4-Autopilot #10779: Use of NOISE param EKF2 and gain tuning

tuloski, 2018-10-29:
I was wondering how the noise params are used in EKF2.
In the code it seems they are not used at all.
So my next question is if there is a way to "tune" the EKF, i.e. for example to give more weight to a certain sensor (for example I have a RTK GPS or a very precise altimeter). Also that seems to be missing atm, or did I misread the code?

hamishwillee, 2018-10-29:
Does this help: http://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html#drag-specific-forces

hamishwillee, 2018-10-29:
@priseborough FYI

mhkabir, 2018-10-29:
They're not documented well, but generally reducing the noise parameters will cause the EKF to trust that sensor more.

Rest assured that all the parameters are used in the code.

tuloski, 2018-10-29:
@mhkabir what it means "generally"? :)
Usually in KF the noise covariance is used to calculate the gains of the filter but here they are not intended as covariance for that afaik.
Can you point me at the code where those parameters are used? For example to calculate the gains for the innovation? I didn't find anything like that...

mhkabir, 2018-10-30:
https://github.com/PX4/ecl/blob/master/EKF/covariance.cpp

Please use discuss.px4.io for further questions.

closed 2018-10-30 by mhkabir