PX4/PX4-Autopilot #10782: add support for MAV_CMD_DO_MOTOR_TEST (GCS driven motor testing)

dagar, 2018-10-29:
A semi-frequent request is to support testing individual motors from the ground. We already have most of the pieces to do this both in QGC and PX4 side (motor_test uORB + cli). I'd like to find a way to fully implement this safely.

https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOTOR_TEST
[image]

Questions
 - how should this work wrt arming?
 - how do a user abort?
 - what if the GCS connection is lost?

dagar, 2018-10-29:
Proposal:
 - add new arming state for motor test
 - output modules (px4io, px4fmu, uavcan, etc) respect arming state
 - in arming state motor test output modules ignore regular `actuator_controls` path and subscribe to `test_motor` https://github.com/PX4/Firmware/blob/master/msg/test_motor.msg
 - safety button should be able to abort motor test
 - any GCS timeout should abort motor test
 - mavlink report armed to GCS?
 - GCS should be able to disarm to kill motor test

dagar, 2018-10-29:
Use same mechanism for testing regular servos?

dagar, 2018-10-29:
QGC side:
[image]

xdwgood, 2018-10-30:
Proposal:
I think `add new arming stateis` not the best way to test the motor, which leads us to change some output modules. We can set a motor test enable parameter. If this parameter is 1, we will directly transfer the received motor speed to the ESC.
param set 1--->unlock ESC--->received test_motor.msg--->actuator execution :smiley: 

I have two question: We test the motor just to prove that it can be rotated? Or to test the motor performance at different speeds? What does GCS use to determine the motor test pass?

Others（Add a stop test button and lose the GCS link to stop） I feel good.

dagar, 2018-10-30:
It's primarily intended as a setup helper. Checking basic things like motor ordering and direction.

Handling it with a new arming state would require changing the output modules, but so would using parameters. The reason to use the arming state machine is because that's what the output modules already respect. We can prevent real arming and motor test from interfering and have well defined transitions.

DonLakeFlyer, 2018-11-08:
Any progress: I'm at the point in a new build where I need to do motor testing. I could get this all wrung out end to end.

dagar, 2019-01-21:
Related discussion - https://github.com/PX4/Firmware/pull/9544#issuecomment-444832994

booo, 2020-03-27:
Can someone summarize the current state of implementation?

hamishwillee, 2020-03-29:
@booo I'm pretty sure this is complete - done in another PR: https://docs.px4.io/master/en/config/motors.html

@bkueng Can this be closed?

bkueng, 2020-03-30:
Yes closing, but would be nice to have the AUX pins as well.

closed 2020-03-30 by bkueng

hamishwillee, 2020-03-30:
@bkueng Create a separate issue to track that? Or is that just going to end up stale?

bkueng, 2020-03-31:
> Or is that just going to end up stale?

Yes I think it would.