PX4/PX4-Autopilot #10783: Telemetry do not work with FrSky R9 series receivers with Pixracer R14

maroquio, 2018-10-30:
After a long time testing with different parameters and configurations, I've concluded that something is buggy when sending telemetry to FrSky R9 series receivers from a Pixracer R14. I've tried to get telemetry with 3 different models of FrSky receivers: **1) X4R-SB**; **2) R9 Slim**; and **3) R9 Mini**.

With receiver 1, everything worked perfectly with both "Arducopter 3.6 stable" and "PX4 1.8.1". The same did not happen with receivers 2 and 3. The radio simply did not discover the sensors, only RSSI and RxBat. To be sure the receivers 2 and 3 are okay, I've tested them using Navio2/RPi3 flight controller running "Arducopter 3.6 stable", and it worked perfectly with full telemetry. So, please, I kindly ask you to check why these R9 series don't work, because they are excellent options for long range flights and I need it working for academic purposes.

If you need any other info, just ask and I will answer as fast as I can. Thanks!

sparky002, 2018-10-30:
I haven't used either of those receivers but I have used the larger R9 on the Pixracer R14 connected to a R9M. I don't get RSSI with it though. I couldn't get telemetry for quite awhile due to having my power level set above 100mW on my transmitter. FrSky released newer firmware recently, called Flex that allows telemetry at all power levels and my issue was fixed. 
This doesn't seem to be your issue though given telemetry works on arducopter. Have you tried the most recent px4 daily build from the git repo?

closed 2019-02-12 by stale[bot]

rolyexpress, 2019-03-08:
So only R9 standard works with telem?

maroquio, 2019-03-08:
The **R9 MM** receiver works with telemetry. Good luck!