PX4/PX4-Autopilot #10784: Making UART ttys3 port available for Rangefinder in Pixhawk4 (FMUv5)

nikan8, 2018-10-30:
Hello everyone, 
has anyone tried making ttyS3 port (UART&I2CB socket) available for an arbitrary serial device? Currently, it is hardcoded as Mavlink and/or secondary GPS use. Attempting to start the rangefinder (e.g. tfmini) driver on ttys3 produces glitches during mag0 initialization. 
any advice appreciated. thank you.

mhkabir, 2018-10-30:
@dagar This is what we were talking about.

dagar, 2018-10-30:
Ok, we just need to figure out what to call it in the generated parameters.

On a pixhawk 4 it's `UART & I2C B`. I'm not sure about the Pixhawk 4 mini or other px4fmu-v5 compatible hardware.

@davids5 @bkueng FYI

dagar, 2018-10-30:
@nikan8 how are you starting it? Did you try using configuring telem 4 (TEL4) via parameters?

nikan8, 2018-10-30:
dagar, All I did was remove mavlink startup on ttyS3 in /ROMFS/px4fmu_common/init.d/rcS and starting tfmini driver on this port (all done on v1.8.1)

dagar, 2018-10-30:
Could you give PX4 master a try? The serial devices are now fully configurable via parameter.

nikan8, 2018-10-30:
Just flashed master on the Pixhawk4, however I see no serial config parameters in QGC. Should I start the driver by hardcoding the port in ROMFS?

mhkabir, 2018-10-30:
There is no need to hardcoded anything : https://docs.px4.io/en/peripherals/serial_configuration.html

davids5, 2018-10-30:
@dagar on all FMUv5 compatible boards that is the TELEM4 connector with UART4 and I2C2.

bkueng, 2018-11-05:
I understand that this is resolved then. Please re-open if not.

closed 2018-11-05 by bkueng