PX4/PX4-Autopilot #10790: After mavlink reboot command, all params lost

KeyneZzz, 2018-10-31:
**Describe the bug**
I'm using an offboard computer and a pixhawk2 to control my drone, if the drone is normally landed it will reboot the px4 using MAVROS. 
Once half a year ago, after rebooting, the drone wasn't able to disarm. When checking on QGC, i found the params on px4 all set to default, and the calibrated params all lost. The px4 version was 1.6.2. I've tried to send reboot mavlink command using pymavlink to an isolate px4, letting it reboot again and again, but the problem seems not appearing.
A week ago, the problem appeared again with ver1.8.0. But it was a newly bought pixhawk2. So I'm wondering if a reboot command on mavlink after some long flight would result in this, or just bad flash driver. 
I want to see if anyone is meeting a same or similar problem, so that we can learn more about the problem.

dagar, 2018-11-05:
Was the vehicle armed when you issued a reboot?

KeyneZzz, 2018-11-06:
it's disarmed. after disarmed for about 10 secs it would send a reboot.

deksprime, 2018-11-06:
This always happened to us in the past. Now it doesn't happen. When it happened, it was happening mostly because of: 

1 - A faulty SD card is inserted into Pixhawk
2 - If 5V cable is used on my companion computer connection, which somehow affects the boot of pixhawk when copter is powered on
3 - Something is booting the Pixhawk but not powering the servo rail properly so the ESCs somehow doesn't start properly

hope this helps.

closed 2019-02-18 by stale[bot]