PX4/PX4-Autopilot #10795: No HIL_ACTUATOR_CONTROLS

AymenHS17, 2018-10-31:
I am developing a custom simulator with HITL. Everything is working well when I use PX4 with Pixhawck (2.4.8) hardware. I can send sensor and command messages to the controller and I can receive HIL_ACTUATOR_CONTROLS, but when I put the same PX4 firmware (1.7.0 and above) on PixRacer I do not receive the HIL_ACTUATOR_CONTROLS message. I am receiving all other messages (HEARTBEAT, EXTENDED_SYS_STATE, etc.) and the controller is responding to my Arming commands but there is no HIL_ACTUATOR_CONTROLS.

HIL_MODE is enabled. I am using the HIL Quadcopter X config. COM_RC_IN_MODE is set to "Joystick/No RC Checks". NAV_DLL_ACT is set to "Disabled".

Any Help ? Thank you.

rligocki, 2018-11-09:
May I ask you how did you get it running on pixhawk 2.4.8 ? I also develop custom simulator with HITL, but I am not able to get any output actuators. I already read code of HIL_ACTUATOR_CONTROLS, but I didn't found any bug which would cause this kind of problem. Interesting is, that you have problem with different hardware even when data comes from actuator_outputs topic.

rligocki, 2018-11-14:
Maybe I found problem. Look at nuttx_px4fmu-v4_default.cmake and uncomment pwm_out_sim driver. I had this problem and this was that problem

AymenHS17, 2018-11-14:
Thank for your reply. If you mean the line in this file (https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v4_default.cmake ), it is already uncommented.

I have commented sensors reset as well following the recommandations here (https://github.com/PX4/Firmware/issues/5651) but still have the same problem.

For Pixhawk 2.4.8 (FMU V2) you need to choose a HIl config (e.i. HIL Quadcopter X). Make sure HIL is enabled. Start sending "Hearbeat" and "Arm" command. You will start receiving HIL_ACTUATOR_CONTROLS as soon as the controller is armed.

Hope it will help.

rligocki, 2018-11-14:
I have it already working. I found out, that there was commented pwm_out_sim in nuttx cmake file for FMU V2.

AymenHS17, 2018-11-14:
Glad to hear that. I confirm it is commented for FMU-V2 but it is working for me. However it is uncommmented for FMU-V4 (Pixracer) and it is not working !

closed 2019-02-26 by stale[bot]