PX4/PX4-Autopilot #10797: UAVCAN config

Avysuarez, 2018-11-01:
@pavel-kirienko @mhkabir 
I working with a new build, with UAVCAN ESC and GPS of zubax, I already all connected and I made the configuration of the ESC, the copter arming but I can't configure the order of the motors, because I have a problem with the uavcan motor assignment, when I press the button to start the assignment the pixhawk does not anything. 
[image]
this is the message that QGC shows me. 
Already tried with the different options of the UAVCAN_ENABLE parameter and nothing. 
https://www.youtube.com/watch?v=RFgwBKcr1qU
what could be the problem?

mhkabir, 2018-11-01:
Are you using Myxas? If so, they don't support automatic enumeration. Slack me and I can guide to through the manual setup process.

bresch, 2018-11-06:
@Avysuarez Here's a good document to read: https://forum.zubax.com/t/quick-start-guide-for-myxa-v0-1/911

Avysuarez, 2018-11-13:
I have the same issue, already followed all the instruction that @mhkabir gave me, and the vehicle only flips on the takeoff. The assignments and the rotation directions of the motors it's correct. I don't know what else could it be. 
This is a video of the ESC #1 
[image]
[image]

closed 2018-11-13 by Avysuarez