PX4/PX4-Autopilot #10808: Esc calibraton command broken?

RomanBapst, 2018-11-05:
I recently tried using the esc_calib app on a Pixracer (fmu v4) and did not have success.

After killing the mc_attitude_controller this is how is started the calibration:
`esc_calib -d /dev/pwm_output0 -c 1234 -l 1000 -h 2000`

After being told to connect the battery and wait for the first ESC indication (high pwm set) the console froze and the ESC did not respond to anything.
I was using PX4 stable 1.8.

YuCodes, 2020-01-21:
I met with the same thing with px4_fmuv3, it asks to hit ENTER and the console freezes. 
[image]

julianoes, 2020-01-24:
@YuCodes which version have you flashed?

YuCodes, 2020-01-24:
I tried it with v1.10

FlavioTonelli, 2020-01-24:
I faced this issue some time ago. It seems not related to PX4 but instead to QGC. It is not correctly recognizing the "Enter" input character.

FlavioTonelli, 2020-01-31:
This is happening because QGC is filtering the "Enter" character. The GUI is using Enter as a confirmation and its therefore don't sending it over the NSH console.
@julianoes i see two ways to fix this, either changing the character on the px4 side (from Enter to a different one , 'f' for example) or change the input logic on QGC side.

julianoes, 2020-02-12:
Aha @FlavioTonelli! @bkueng you might have an idea here.

yuthikasagarage, 2020-02-12:
I tried it using another key without using 13 at esc_calib/esc_calib.c file, and it does work as @FlavioTonelli said.
			if (c == **13**) {
				break;