PX4/PX4-Autopilot #10810: git submodule update fails for PX4 Firmware 1.6.5

AbhishekGS-MW, 2018-11-06:
**Describe the bug**
The git submodule update fails for PX4 Firmware 1.6.5 with the following error:

_fatal: repository 'https://github.com/PX4-NuttX/NxWidgets.git/' not found
Clone of 'https://github.com/PX4-NuttX/NxWidgets.git' into submodule path 'NxWidgets' failed
Failed to recurse into submodule path 'NuttX'_

Subsequently, the build also fails (make px4fmu-v2_default) with the following error:

_1 out of 1 hunk ignored
make[3]: *** [px4fmu-v2/NuttX/px4fmu-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch.stamp] Error 1
make[2]: *** [CMakeFiles/px4fmu-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs...._

**To Reproduce**
Steps to reproduce the behavior:
1. Open Ubuntu Bash on Windows 10
2. Clone the PX4 Firmware ( git clone https://github.com/PX4/Firmware.git Firmware)
3. Checkout 'v1.6.5' ( cd Firmware;
git checkout v1.6.5 )
4. git submodule update --init --recursive

**Expected behavior**
A successful PX4 firmware build for v1.6.5

**Additional context**

I suspect the below change could be the issue.
[https://github.com/PX4/Firmware/commit/d61fbbf5680d4f4dbc1e523c00d0aa8babfa2212](https://github.com/PX4/Firmware/commit/d61fbbf5680d4f4dbc1e523c00d0aa8babfa2212
)

dagar, 2018-11-09:
@AbhishekGS-MW this has been fixed, please try again. The NxWidgets repository is now archived at https://github.com/PX4/NuttX-NxWidgets with appropriate redirects configured.

I'd still strongly recommend you update to the current stable PX4 v1.8.1.

closed 2018-11-09 by dagar

dagar, 2018-11-13:
@AbhishekGS-MW I'll restore (and transfer) that one as well. For now you can simply delete it.

AbhishekGS-MW, 2018-11-15:
@dagar The _git submodule update --init --recrsive_ goes through successfully, however the build (_make px4fmu-v2_default_) fails with the following error.

_[ 14%] Building NuttX for px4io-v2 with nsh
make[4]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
stdio/lib_sscanf.c: In function 'vsscanf':
stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[5]: *** [bin/lib_sscanf.o] Error 1
make[4]: *** [libc/libc.a] Error 2
make[3]: *** [px4io-v2/nuttx_export.stamp] Error 2
make[2]: *** [src/modules/px4iofirmware/CMakeFiles/nuttx_export_px4io-v2.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
stdio/lib_sscanf.c: In function 'vsscanf':
stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[5]: *** [bin/lib_sscanf.o] Error 1
make[4]: *** [libc/libc.a] Error 2
make[3]: *** [px4fmu-v2/nuttx_export.stamp] Error 2
make[2]: *** [CMakeFiles/nuttx_export_px4fmu-v2.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [px4fmu-v2_default] Error 2_

dagar, 2018-11-15:
That's a gcc version issue. Which version of arm-none-eabi-gcc are you using?

AbhishekGS-MW, 2018-11-16:
@dagar , thanks for the clue. I was initially using GCC 7.2. Your comment helped me realize that I needed GCC 5.4 version and it now builds successfully with GCC 5.4. Thank you.