PX4/PX4-Autopilot #10813: Bug of HITL simulation

roseyangyu, 2018-11-07:
I have test HILT simulation with gazebo.I have done following [Dev setup](http://dev.px4.io/en/simulation/hitl.html).But I run "gazebo Tools/sitl_gazebo/worlds/iris.world" after connected the flight controller.This is the issues in followwing picture.
[image]
Then I test jmavsim.I run "./Tools/jmavsim_run.sh -q -d /dev/ttyACM0 -b 921600 -r " and connect flight controller and QGC.When armed,it messy fly and doesn't fly according to the mission in QGC.But it has not moved in jmavsim.
[image]
[image]
`[15:49:11.649] Critical: Takeoff denied, disarm and re-try
[15:49:31.652] Critical: REJECT AUTO HOLD
[15:49:37.713] Info: [logger] file: /fs/microsd/log/2018-11-07/07_49_3
[15:49:37.915] Info: Takeoff detected
[15:49:37.916] EMERGENCY: CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR
[15:49:39.945] Info: Landing detected
[15:49:41.554] Info: Takeoff detected
[15:49:46.096] Critical: Failsafe enabled: no global position
[15:49:46.097] Info: Failsafe mode enabled
[15:49:46.097] Info: Executing Mission
[15:49:46.097] Info: Takeoff to 50.0 meters above home.
[15:49:47.495] Info: Failsafe mode disabled
[15:49:47.495] Info: Takeoff to 50.0 meters above home.
[15:49:47.495] Critical: Failsafe enabled: no global position
[15:49:47.496] Info: Failsafe mode enabled
[15:49:48.883] Info: Failsafe mode disabled
[15:49:48.884] Info: Takeoff to 50.0 meters above home.
[15:49:48.884] Critical: Failsafe enabled: no global position
[15:49:48.884] Info: Failsafe mode enabled
[16:19:40.637] Critical: MANUAL CONTROL LOST (at t=2012522ms)
[16:19:46.077] Info: MANUAL CONTROL REGAINED after 5872ms
[16:19:54.773] Critical: MANUAL CONTROL LOST (at t=2026626ms)
[16:19:58.796] Info: MANUAL CONTROL REGAINED after 4449ms
[16:20:19.826] Critical: NOT DISARMING: Not in manual mode or landed yet.
[16:20:21.241] Critical: NOT DISARMING: Not in manual mode or landed yet.
[16:20:22.651] Critical: NOT DISARMING: Not in manual mode or landed yet.
[16:20:23.854] Critical: NOT DISARMING: Not in manual mode or landed yet.
[16:20:27.096] Critical: NOT DISARMING: Not in manual mode or landed yet.
[16:20:28.905] Critical: REJECT AUTO HOLD
[16:20:29.115] Info: Failsafe mode disabled
[16:21:05.290] Info: Takeoff to 50.0 meters above home.
[16:21:05.292] Critical: Failsafe enabled: no global position
[16:21:05.293] Info: Failsafe mode enabled
[16:21:08.123] Info: Failsafe mode disabled
[16:21:08.123] Critical: Failsafe enabled: no global position
[16:21:08.124] Info: Failsafe mode enabled
[16:21:08.124] Info: Takeoff to 50.0 meters above home.
[16:21:28.927] Critical: MANUAL CONTROL LOST (at t=2120797ms)
[16:41:23.872] Critical: Set SYS_HITL to 0 and reboot to disable HITL.
[16:41:24.265] Info: data link #0 lost
[16:41:24.336] Critical: ALL DATA LINKS LOST
[16:41:24.436] Info: data link #0 regained`
this is log in QGC

roseyangyu, 2018-11-07:
My flight controller is px4v2.FW version is 1.8.0 releases

julianoes, 2018-11-07:
What is the FPS number in the bottom left of jMAVSim? It should be around 60 Hz. If it drops to 5 Hz something is wrong and you need to restart or try to lower the rate to e.g. `-r 100`.

roseyangyu, 2018-11-07:
[image]
[image]

roseyangyu, 2018-11-07:
this is I try -r 100

roseyangyu, 2018-11-07:
They can't move all

julianoes, 2018-11-08:
@roseyangyu and you set `SYS_HITL` to `1`?

roseyangyu, 2018-11-08:
yes,I have set that.I have connected flight controller and gazebo.But it can't fly same to jmavsim.It doesn't move in any circumstances,in mission mode or manual mode.It can be armed.Speed and alttude-rel is zero aways.As gazebo,v1.8.0 is not connected with flight controller,but v1.7.0 can be.

roseyangyu, 2018-11-09:
@julianoes 
new version is not support px4-v2 and v3,I use v4 is successful!

julianoes, 2018-11-09:
>  new version is not support px4-v2 and v3,I use v4 is successful!

Hm, that's interesting, we need to figure out why.

lovettchris, 2018-11-28:
Do you see hilactuator_controls messages in the mavlink stream?

hamishwillee, 2018-11-29:
@julianoes Is there a bug for tracking this issue? ie this is closed, but you say in https://github.com/PX4/Firmware/issues/10813#issuecomment-437289551 that there are still problems.

julianoes, 2018-11-29:
> new version is not support px4-v2 and v3,I use v4 is successful!

That's all the info I have. I suppose we can leave it closed until we find that something doesn't work again. I'll remember to try HITL again sometime.

roseyangyu, 2018-11-29:
I have queried the same problem in [https://github.com/PX4/Firmware/issues](url) and [http://discuss.px4.io/](url). Someone test v2 to HITL is failed,but pixhawk 4 is successful use v1.7.0.And v2 and ppixhawk 4 are failed

ks0ze, 2019-02-08:
@julianoes, I'm also having this issue with an mRo Pixhawk running the 1.8.2 px4fmu-v3_default version of the firmware. Sometimes I'm able to get a correct take-off and if flies fine for a few seconds then I get a "Failsafe Enabled: no global postion" warning and things start to go crazy like in the initial post here.

The odd thing is I've been able to run this 100% successfully on another machine with the same firmware and hardware, but I unfortunately no longer have access to that machine.

julianoes, 2019-02-11:
@ks0ze when this happens, does the FPS number in jMAVSim go from 60 Hz to about 5 Hz?

I have seen that case where the serial communication is somehow overloaded, so gets unstable and drops to 5 Hz. All that helps in that case is to restart the Pixhawk and HITL, and maybe try another USB port or try without a USB hub, etc.

ks0ze, 2019-02-11:
I was seeing an occasional drop from 60Hz to 20-30Hz but it would immediately jump back up to 60Hz.

I just did a fresh install of the OS (Ubuntu 18.04) and rebuild jmavsim this morning and things seem to be working better for me now but not entirely flawless. Out of about 10 HITL flights, two of them lost global position and went crazy but the rest of the flights completed successfully. However, even on the good flights there seems to be a lot of jitter in the artificial horizon/altitude gauge in QGC.

The rate seems to stay at 60FPS no matter what setting I pass with the -r command (I've tried 30, 60, 100, and 250). I see the same behavior on both USB2 and USB3 ports with no hub.

The computer I was originally working with before experiencing any problems what a much more capable machine that what I'm currently using, so it makes sense something could be overloaded.

julianoes, 2019-02-12:
It's expected that jMAVSim shows 60 Hz because that's what the display/graphics card support. I expect more jitter in attitude if you lower the rate further  down than 250 Hz.

> The computer I was originally working with before experiencing any problems what a much more capable machine that what I'm currently using, so it makes sense something could be overloaded.

Right, that's possible. I had that once and had to go to 100 Hz to have it work somewhat consistently. I also once spent a few days debugging the serial stack of jMAVSim but without success, unfortunately.

julianoes, 2019-07-01:
@roseyangyu closing as stale. If it's still happening for you please re-open and provide as much detail as possible.

closed 2019-07-01 by julianoes

roseyangyu, 2019-07-08:
@julianoes oh，thanks very much