PX4/PX4-Autopilot #10822: Urgent issue with the Camera trigger.

tubeme, 2018-11-09:
I have an urgent issue with one camera - Mapir https://www.mapir.camera/ camera.

Our regular PWM trigger stays at 1500us then it triggers at 1700us. (not the seagulMap)

For this Mapir camera it is no go because at 1500us it switches to USB file transfer mode.

The trigger should stay at 1000us passive and then receive 2000us on trigger events.

Is there any quick fix or workaround I can apply to make it work until 14th of Nov. So I suppose it should be a dirty hack.

Asking for help on this one. Really don't have any clue how to do the survey in a demo mission with this camera. 

Any help will be greatly appreciated.

mhkabir, 2018-11-09:
Two line change : https://github.com/PX4/Firmware/blob/master/src/drivers/camera_trigger/interfaces/src/pwm.cpp#L9-L11

closed 2018-11-09 by mhkabir

tubeme, 2018-11-11:
Hi @mhkabir Is it very hard to make parameters for these two? We are willing to pay somebody to do it for the open source branch. Will you be able to do it? We can discuss in private. vasil.petrov.tv@gmail.com